PX4/PX4-Autopilot #14527: Arming State Machine unexpected cases "Transition denied:"

MaEtUgR, 2020-03-30:
**Describe the bug**
The arming state machine 
https://github.com/PX4/Firmware/blob/73f7243bc8350664864685d1a0adbeaaa2f2c5e5/src/modules/commander/state_machine_helper.cpp#L59-L73
is working and unit tested but in some use cases the state machine can get called with illegal transitions and the user is presented with an unexpected message saying "Transition denied:" from
https://github.com/PX4/Firmware/blob/73f7243bc8350664864685d1a0adbeaaa2f2c5e5/src/modules/commander/state_machine_helper.cpp#L237

This issue tries to track those cases since they seem to unveil common problems.

**To Reproduce**
- `Transition denied: SHUTDOWN to INIT` / `Transition denied: SHUTDOWN to ARMED`
1. Power drone with battery and power module such that it measures a battery voltage and state
2. Plug autopilot to computer via USB
3. Unplug battery
4. The autopilot tries to shutdown but the board doesn't support shutdown (tested on a Pixhawk 4).
5. [image]
6. The autopilot is in SHUTDOWN state and any attempt to calibrate or arm will trigger one of the messages.

- `Transition denied: ARMED to INIT`
1. Arm and start flying the drone
2. Open the Radio tab in QGC
3. 
[image]

This is an issue describing one of the possible cases: https://github.com/PX4/Firmware/issues/14158

**Expected behavior**
The guards for the transitions and the messaging that enables debugging should stay as they are since they prevent further safety-critical problems. But the calls to the state machine should be revised such that it's highly unlikely the user is ever bothered with a transition guard message.

tacdin, 2020-04-10:
I'm having the same problem. I would be glad if you can direct the developers and users. It is very frustrating to get this error. No operations can be performed.

dagar, 2020-05-04:
Should be fixed by https://github.com/PX4/Firmware/pull/14825.

MaEtUgR, 2020-05-06:
It is, see https://github.com/PX4/Firmware/pull/14825#issuecomment-624658878

closed 2020-05-06 by dagar