PX4/PX4-Autopilot #14541: ERROR when make px4_sitl_default gazebo

mjsky97, 2020-03-31:
Hello, i'm just installed ROS kinetic and cloned PX4 Frimware
I already checked 'make px4_sitl_default' command work well
but I try to this with gazebo option, the error causes.

I'm using Ubuntu 16.04 LTS 64 bits, and ROS kinetic and I'm really want to solve this problem

This is log of terminal
[  0%] Built target lockstep_scheduler
[  1%] Generating ../../logs
[  2%] Built target drivers_board
[  2%] Built target uorb_headers
[  2%] Built target logs_symlink
[  2%] Built target mixer_gen
[  3%] Built target mixer_gen_6dof
[  3%] Built target ver_gen
[  4%] Built target git_ecl
[  4%] Built target git_gps_devices
[  4%] Built target git_mavlink_v2
[  4%] Built target git_gazebo
[  5%] Built target tinybson
[  6%] Built target perf
[  8%] Built target work_queue
[  9%] Built target ecl_airdata
[  9%] Built target ecl_geo
[  9%] Built target ecl_geo_lookup
[ 10%] Built target ecl_tecs
[ 10%] Built target ecl_validation
[ 28%] Built target uorb_msgs
[ 28%] Built target Mixer
[ 29%] Built target rc
[ 29%] Built target version
[ 29%] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Using C++14 compiler
[ 30%] Built target parameters
[ 30%] Built target ecl_l1
[ 30%] Built target SimpleMixer
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   filesystem
--   chrono
--   date_time
--   atomic
[ 30%] Built target HelicopterMixer
[ 30%] Built target NullMixer
[ 31%] Built target MultirotorMixer
[ 31%] Built target generate_topic_listener
-- Boost version: 1.58.0
[ 31%] Built target mixer
[ 33%] Built target modules__uORB
[ 34%] Built target px4_work_queue
-- Checking for module 'gstreamer-1.0 >= 1.0'
[ 35%] Built target px4_platform
--   
-- Checking for module 'gstreamer-base-1.0 >= 1.0'
--   
[ 36%] Built target px4_daemon
[ 37%] Built target px4_layer
[ 38%] Built target systemlib
[ 39%] Built target cdev
-- Building klt_feature_tracker without catkin
[ 39%] Built target airspeed
[ 39%] Built target bezier
[ 39%] Built target AirspeedValidator
-- Building OpticalFlow with OpenCV
[ 39%] Built target hysteresis
[ 40%] Built target battery
[ 41%] Built target lib__cdev__test__cdev_test
[ 41%] Built target circuit_breaker
[ 42%] Built target CollisionPrevention
[ 43%] Built target controllib
[ 43%] Built target lib__controllib__controllib_test
[ 44%] Built target conversion
[ 45%] Built target drivers__device
[ 45%] Built target mathlib
[ 45%] Built target drivers_barometer
[ 46%] Built target drivers_magnetometer
[ 46%] Built target FlightTask
[ 47%] Built target landing_slope
[ 47%] Built target mixer_module
[ 47%] Built target pid
[ 47%] Built target output_limit
[ 47%] Built target lib__rc__rc_tests
[ 47%] Built target tunes
-- catkin DISABLED
-- Gazebo version: 7.0
[ 47%] Built target mag_compensation
[ 47%] Built target arch_tone_alarm
[ 48%] Built target WeatherVane
-- Found GStreamer: adding gst_camera_plugin
-- Found GStreamer: adding gst_video_stream_widget
[ 48%] Built target modules__uORB__uORB_tests
[ 50%] Built target lib__work_queue__test__wqueue_test
[ 50%] Built target drivers__camera_capture
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_LIBRARIES (ADVANCED)
    linked by target "LiftDragPlugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gst_camera_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_parachute_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_catapult_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_barometer_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_wind_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_geotagged_images_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_mockup_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_magnetometer_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "sensor_msgs" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_irlock_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_video_stream_widget" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_controller_interface" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "std_msgs" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gps_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_usv_dynamics_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_sonar_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "mav_msgs" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "nav_msgs" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_lidar_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_mavlink_interface" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_uuv_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_vision_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_multirotor_base_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_imu_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gimbal_controller_plugin" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "physics_msgs" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_motor_model" in directory /home/minjoon/catkin_ws/src/Firmware/Tools/sitl_gazebo

-- Configuring incomplete, errors occurred!
See also "/home/minjoon/catkin_ws/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/minjoon/catkin_ws/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
platforms/posix/CMakeFiles/sitl_gazebo.dir/build.make:105: recipe for target 'external/Stamp/sitl_gazebo/sitl_gazebo-configure' failed
make[4]: *** [external/Stamp/sitl_gazebo/sitl_gazebo-configure] Error 1
CMakeFiles/Makefile2:33059: recipe for target 'platforms/posix/CMakeFiles/sitl_gazebo.dir/all' failed
make[3]: *** [platforms/posix/CMakeFiles/sitl_gazebo.dir/all] Error 2
make[3]: *** Waiting for unfinished jobs....
[ 50%] Built target drivers__pwm_out_sim
[ 51%] Built target drivers__gps
[ 52%] Built target drivers__camera_trigger
CMakeFiles/Makefile2:25688: recipe for target 'platforms/posix/CMakeFiles/gazebo.dir/rule' failed
make[2]: *** [platforms/posix/CMakeFiles/gazebo.dir/rule] Error 2
Makefile:7802: recipe for target 'gazebo' failed
make[1]: *** [gazebo] Error 2
Makefile:198: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 2

julianoes, 2020-03-31:
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_LIBRARIES (ADVANCED)
```

https://github.com/PX4/Firmware/blob/d609639367eaecb59dcb322493235a8c552c4e03/Tools/setup/ubuntu.sh#L190-L196

closed 2020-03-31 by julianoes

mjsky97, 2020-03-31:
> ```
> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
> Please set them or make sure they are set and tested correctly in the CMake files: [...]

Thank you for your help :-)

julianoes, 2020-03-31:
@mjsky97 no worries!