PX4/PX4-Autopilot #14547: fmu-v4_cannode stuck in boot loop

JacobCrabill, 2020-03-31:
Sometime recently the cannode build target for the Pixracer (fmu-v4) was broken.  It will boot, but upon connecting a Zubax Babel board to the CAN port, the Babel will show that the cannode gets stuck in an infinite reboot loop.  It will briefly show up as initializing, then power off and disappear from the UAVCAN GUI, show up again as initializing, etc.

This commit still works: [pull/14444/d14fc944](https://github.com/PX4/Firmware/pull/14444/commits/d14fc944c0a1eb185173fd607e429bb5178ce85b)

But as of last Friday at least (and I tried yesterday as well) master does not work.

@dakejahl

dakejahl, 2020-03-31:
Can you describe your test setup? I just tested this on a Pixracer with a GPS attached and everything works as expected.

JacobCrabill, 2020-03-31:
Hmm.  Using commit [2b82b471](https://github.com/PX4/Firmware/commit/2b82b471c1ec4baf9fa4aeacb6788bf0109c1db4)? I'll give that another try.

dakejahl, 2020-03-31:
Yup whatever the latest commit is

JacobCrabill, 2020-03-31:
Still doesn't work for me; I just get this in the UAVCAN GUI when connected to a Babel:
[image]

It stays in the "INITIALIZATION" mode for a second or two then goes away.  And now after a few minutes, it's no longer rebooting itself at all is doesn't appear on the CAN bus at all.

Using an mRobotics Pixracer; brand-new purchase from last week.

JacobCrabill, 2020-03-31:
Oh, interesting.  I removed the micro-USB cable, replug the CAN cable, and now it works 🤔

JacobCrabill, 2020-03-31:
Somewhere the handling of USB power on this must have changed.  All of my prior testing has been with just leaving the USB cord plugged in.

At any rate, it's looking good now!
[image]

dakejahl, 2020-03-31:
Do you have a FTDI to hook into the console? The USB cable being plugged in should make no difference, I've got mine plugged in with a USB as well.

JacobCrabill, 2020-03-31:
I do; which connector/pins expose the console on the Pixracer?

dakejahl, 2020-03-31:
It's part of the pixhawk debug port
https://dev.px4.io/v1.9.0/en/debug/system_console.html

JacobCrabill, 2020-03-31:
Ah, I see it.  I have a DroneCode Probe; I'll give that a try later when I have time.

JacobCrabill, 2020-04-02:
Well I rebased @dakejahl's airspeed branch on top of master (actually this branch here: https://github.com/volans-i/Firmware/tree/pr-uavcan-airspeed) and it all works now 🤷‍♂️
[image]

closed 2020-04-02 by JacobCrabill