PX4/PX4-Autopilot #14554: Problem when running 'make px4_sitl jmavsim' error when generating uORB topic headers FAILED:

Bionic-Sock, 2020-03-31:
Hello! I'm really new to all this and I could use some help. Thank you very much in advance.

I'm using Ubuntu 18.04 and am having a huge problem when I try to run '_make px4_sitl jmavsim_'.

I followed these instructions:
 [https://dev.px4.io/v1.9.0/en/setup/building_px4.html]

I've also installed Qground control with no problem and followed the steps for the development toolchain here:
[https://dev.px4.io/v1.9.0/en/setup/dev_env_linux.html]

But now when I try and do the first build I get this output:

**isaac@Cougar-Desktop:~/Firmware$** make px4_sitl jmavsim

-- PX4 version: v1.11.0-beta1-1153-g39b47e63b5
-- PX4 config file: /home/isaac/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-244-g870ec38
-- Configuring done
-- Generating done
-- Build files have been written to: /home/isaac/Firmware/build/px4_sitl_default
[0/730] git submodule src/drivers/gps/devices
[7/730] git submodule src/lib/ecl
[8/730] Generating uORB topic headers
FAILED: uORB/topics/uORBTopics.hpp uORB/topics/actuator_armed.h uORB/topics/actuator_controls.h uORB/topics/actuator_outputs.h uORB/topics/adc_report.h uORB/topics/airspeed.h uORB/topics/airspeed_validated.h uORB/topics/battery_status.h uORB/topics/camera_capture.h uORB/topics/camera_trigger.h uORB/topics/cellular_status.h uORB/topics/collision_constraints.h uORB/topics/collision_report.h uORB/topics/commander_state.h uORB/topics/cpuload.h uORB/topics/debug_array.h uORB/topics/debug_key_value.h uORB/topics/debug_value.h uORB/topics/debug_vect.h uORB/topics/differential_pressure.h uORB/topics/distance_sensor.h uORB/topics/ekf2_timestamps.h uORB/topics/ekf_gps_drift.h uORB/topics/ekf_gps_position.h uORB/topics/esc_report.h uORB/topics/esc_status.h uORB/topics/estimator_innovations.h uORB/topics/estimator_sensor_bias.h uORB/topics/estimator_status.h uORB/topics/follow_target.h uORB/topics/geofence_result.h uORB/topics/gps_dump.h uORB/topics/gps_inject_data.h uORB/topics/home_position.h uORB/topics/hover_thrust_estimate.h uORB/topics/input_rc.h uORB/topics/iridiumsbd_status.h uORB/topics/irlock_report.h uORB/topics/landing_gear.h uORB/topics/landing_target_innovations.h uORB/topics/landing_target_pose.h uORB/topics/led_control.h uORB/topics/log_message.h uORB/topics/logger_status.h uORB/topics/manual_control_setpoint.h uORB/topics/mavlink_log.h uORB/topics/mission.h uORB/topics/mission_result.h uORB/topics/mount_orientation.h uORB/topics/multirotor_motor_limits.h uORB/topics/obstacle_distance.h uORB/topics/offboard_control_mode.h uORB/topics/onboard_computer_status.h uORB/topics/optical_flow.h uORB/topics/orb_test.h uORB/topics/orb_test_large.h uORB/topics/orb_test_medium.h uORB/topics/orbit_status.h uORB/topics/parameter_update.h uORB/topics/ping.h uORB/topics/position_controller_landing_status.h uORB/topics/position_controller_status.h uORB/topics/position_setpoint.h uORB/topics/position_setpoint_triplet.h uORB/topics/power_button_state.h uORB/topics/power_monitor.h uORB/topics/pwm_input.h uORB/topics/qshell_req.h uORB/topics/qshell_retval.h uORB/topics/radio_status.h uORB/topics/rate_ctrl_status.h uORB/topics/rc_channels.h uORB/topics/rc_parameter_map.h uORB/topics/rpm.h uORB/topics/safety.h uORB/topics/satellite_info.h uORB/topics/sensor_accel.h uORB/topics/sensor_accel_fifo.h uORB/topics/sensor_accel_integrated.h uORB/topics/sensor_accel_status.h uORB/topics/sensor_baro.h uORB/topics/sensor_combined.h uORB/topics/sensor_correction.h uORB/topics/sensor_gyro.h uORB/topics/sensor_gyro_fifo.h uORB/topics/sensor_gyro_integrated.h uORB/topics/sensor_gyro_status.h uORB/topics/sensor_mag.h uORB/topics/sensor_preflight.h uORB/topics/sensor_selection.h uORB/topics/servorail_status.h uORB/topics/subsystem_info.h uORB/topics/system_power.h uORB/topics/task_stack_info.h uORB/topics/tecs_status.h uORB/topics/telemetry_status.h uORB/topics/test_motor.h uORB/topics/timesync.h uORB/topics/timesync_status.h uORB/topics/trajectory_bezier.h uORB/topics/trajectory_waypoint.h uORB/topics/transponder_report.h uORB/topics/tune_control.h uORB/topics/uavcan_parameter_request.h uORB/topics/uavcan_parameter_value.h uORB/topics/ulog_stream.h uORB/topics/ulog_stream_ack.h uORB/topics/vehicle_acceleration.h uORB/topics/vehicle_air_data.h uORB/topics/vehicle_angular_acceleration.h uORB/topics/vehicle_angular_velocity.h uORB/topics/vehicle_attitude.h uORB/topics/vehicle_attitude_setpoint.h uORB/topics/vehicle_command.h uORB/topics/vehicle_command_ack.h uORB/topics/vehicle_constraints.h uORB/topics/vehicle_control_mode.h uORB/topics/vehicle_global_position.h uORB/topics/vehicle_gps_position.h uORB/topics/vehicle_imu.h uORB/topics/vehicle_land_detected.h uORB/topics/vehicle_local_position.h uORB/topics/vehicle_local_position_setpoint.h uORB/topics/vehicle_magnetometer.h uORB/topics/vehicle_odometry.h uORB/topics/vehicle_rates_setpoint.h uORB/topics/vehicle_roi.h uORB/topics/vehicle_status.h uORB/topics/vehicle_status_flags.h uORB/topics/vehicle_trajectory_bezier.h uORB/topics/vehicle_trajectory_waypoint.h uORB/topics/vtol_vehicle_status.h uORB/topics/wheel_encoders.h uORB/topics/wind_estimate.h uORB/topics/yaw_estimator_status.h 
cd /home/isaac/Firmware/msg && /usr/bin/python3 tools/px_generate_uorb_topic_files.py --headers -f actuator_armed.msg actuator_controls.msg actuator_outputs.msg adc_report.msg airspeed.msg airspeed_validated.msg battery_status.msg camera_capture.msg camera_trigger.msg cellular_status.msg collision_constraints.msg collision_report.msg commander_state.msg cpuload.msg debug_array.msg debug_key_value.msg debug_value.msg debug_vect.msg differential_pressure.msg distance_sensor.msg ekf2_timestamps.msg ekf_gps_drift.msg ekf_gps_position.msg esc_report.msg esc_status.msg estimator_innovations.msg estimator_sensor_bias.msg estimator_status.msg follow_target.msg geofence_result.msg gps_dump.msg gps_inject_data.msg home_position.msg hover_thrust_estimate.msg input_rc.msg iridiumsbd_status.msg irlock_report.msg landing_gear.msg landing_target_innovations.msg landing_target_pose.msg led_control.msg log_message.msg logger_status.msg manual_control_setpoint.msg mavlink_log.msg mission.msg mission_result.msg mount_orientation.msg multirotor_motor_limits.msg obstacle_distance.msg offboard_control_mode.msg onboard_computer_status.msg optical_flow.msg orb_test.msg orb_test_large.msg orb_test_medium.msg orbit_status.msg parameter_update.msg ping.msg position_controller_landing_status.msg position_controller_status.msg position_setpoint.msg position_setpoint_triplet.msg power_button_state.msg power_monitor.msg pwm_input.msg qshell_req.msg qshell_retval.msg radio_status.msg rate_ctrl_status.msg rc_channels.msg rc_parameter_map.msg rpm.msg safety.msg satellite_info.msg sensor_accel.msg sensor_accel_fifo.msg sensor_accel_integrated.msg sensor_accel_status.msg sensor_baro.msg sensor_combined.msg sensor_correction.msg sensor_gyro.msg sensor_gyro_fifo.msg sensor_gyro_integrated.msg sensor_gyro_status.msg sensor_mag.msg sensor_preflight.msg sensor_selection.msg servorail_status.msg subsystem_info.msg system_power.msg task_stack_info.msg tecs_status.msg telemetry_status.msg test_motor.msg timesync.msg timesync_status.msg trajectory_bezier.msg trajectory_waypoint.msg transponder_report.msg tune_control.msg uavcan_parameter_request.msg uavcan_parameter_value.msg ulog_stream.msg ulog_stream_ack.msg vehicle_acceleration.msg vehicle_air_data.msg vehicle_angular_acceleration.msg vehicle_angular_velocity.msg vehicle_attitude.msg vehicle_attitude_setpoint.msg vehicle_ [... cut ...] e/build/px4_sitl_default/msg/tmp/headers -q
Traceback (most recent call last):
  File "tools/px_generate_uorb_topic_files.py", line 46, in <module>
    from packaging import version
ModuleNotFoundError: No module named 'packaging'
[15/730] Generating git version header
ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

Steps to reproduce the behavior:
1. cd Firmware
2. make px4_sitl jmavsim
3. See error

Thank you for any help

julianoes, 2020-04-01:
```
    from packaging import version
ModuleNotFoundError: No module named 'packaging'
```

Try: `pip3 install --user packaging`.

closed 2020-04-01 by TSC21

Bionic-Sock, 2020-04-01:
You're the man. Thanks a bunch. I also found I had to run the ubuntu.sh script that helped everything.

psp316r, 2020-05-19:
I have the same Issue:

admin1@ubuntu-dtlbhs0011:~/Desktop/Firmware$ make px4_sitl gazebo
-- PX4 version: v1.11.0-beta2-23-g4f82f028e7
-- PX4 config file: /home/admin1/Desktop/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-286-g24f2e60

That's it Jmavsim is not Launching after that

psp316r, 2020-05-20:
I got it Fixed, since the sub modules are missing because of
`git clone https://github.com/PX4/Firmware.git`

So I have used --recursive with it.
`git clone https://github.com/PX4/Firmware.git --recursive`
 which fixed it.

PR #14556 (2020-04-01): tools: present nicer error for missing packaging
This makes the error more user friendly and suggests what to do to fix it.

@TSC21 I beat you to it :smile:.

Fixes #14554.