PX4/PX4-Autopilot #14558: the breathe led did not work during uavcan enable

liganggg, 2020-04-01:
firmware version 1.10.2 ,hardware fmuv5-nano

when i set param UAVCAN_ENABLE = 1,then reboot vehicle ,the breathe led did not work.
and in other case,reset param to zero, led is normal. but after “uavcan start” in nsh,　the led off immediately.
 

thanks~

julianoes, 2020-04-01:
FYI @pavel-kirienko

pavel-kirienko, 2020-04-01:
Let's give your boot logs a look, I suppose.

liganggg, 2020-04-02:
> Let's give your boot logs a look, I suppose.

`nsh> reboot
WARN  [shutdown] Reboot NOW.
[boot] Rev 0x0 : Ver 0x6 V560
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V5
HW type: V560
HW version: 0x00000006
HW revision: 0x00000000
FW git-hash: 1d6c8a439dfebdf397e0acc5917979acc321b858
FW version: 1.10.1 0 (17432832)
FW git-branch: develop_v5
OS: NuttX
OS version: Release 7.29.0 (119341311)
OS git-hash: 427238133be2b0ecd068a11e886ee8fdbc31f6dc
Build datetime: Mar 30 2020 22:08:16
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 25]
PX4GUID: 000200000000383937383138510200380033
MCU: STM32F76xxx, rev. Z
[hardfault_log] Fault Log is Armed
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is
WARN  [rgbled] no RGB led on bus #1
WARN  [rgbled_ncp5623c] no RGB led on bus #1
Board sensors: /etc/init.d/rc.board_sensors
MPU6000 on SPI bus 1 at 1 (1000 KHz)
MPU6000 on SPI bus 1 at 0 (1000 KHz)
BMI055_ACCEL on SPI bus 1 at 3 (10000 KHz)
BMI055_GYRO on SPI bus 1 at 2 (10000 KHz)
INFO  [ist8310] no device on bus 1
INFO  [ist8310] no device on bus 2
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [hmc5883] no device on bus 2 (type: 2)
WARN  [hmc5883] no device on bus 4 (type: 2)
WARN  [qmc5883] no device on bus 1 (type: 2)
WARN  [qmc5883] no device on bus 2 (type: 2)
WARN  [qmc5883] no device on bus 4 (type: 2)
WARN  [lis3mdl] no device on bus 2
WARN  [lis3mdl] no device on bus 2
WARN  [lis3mdl] no device on bus 2
nsh: icm20948: command not found
WARN  [ak09916] AK09916 mag init failed
ERROR [ak09916] driver start failed
IST8310 on I2C bus 3 at 0x0e (bus: 100 KHz, max: 400 KHz)
MS5611_SPI on SPI bus 4 at 0 (20000 KHz)
ERROR [battery_status] no ADC found: /dev/adc0 (2)
INFO  [commander] Mission #2 loaded, 7 WPs, curr: 0
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS0
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
`

`nsh> reboot
WARN  [shutdown] Reboot NOW.
[boot] Rev 0x0 : Ver 0x6 V560
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V5
HW type: V560
HW version: 0x00000006
HW revision: 0x00000000
FW git-hash: 1d6c8a439dfebdf397e0acc5917979acc321b858
FW version: 1.10.1 0 (17432832)
FW git-branch: develop_v5
OS: NuttX
OS version: Release 7.29.0 (119341311)
OS git-hash: 427238133be2b0ecd068a11e886ee8fdbc31f6dc
Build datetime: Mar 30 2020 22:08:16
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 25]
PX4GUID: 000200000000383937383138510200380033
MCU: STM32F76xxx, rev. Z
[hardfault_log] Fault Log is Armed
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is
WARN  [rgbled] no RGB led on bus #1
WARN  [rgbled_ncp5623c] no RGB led on bus #1
Board sensors: /etc/init.d/rc.board_sensors
MPU6000 on SPI bus 1 at 1 (1000 KHz)
MPU6000 on SPI bus 1 at 0 (1000 KHz)
BMI055_ACCEL on SPI bus 1 at 3 (10000 KHz)
BMI055_GYRO on SPI bus 1 at 2 (10000 KHz)
INFO  [ist8310] no device on bus 1
INFO  [ist8310] no device on bus 2
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [hmc5883] no device on bus 2 (type: 2)
WARN  [hmc5883] no device on bus 4 (type: 2)
WARN  [qmc5883] no device on bus 1 (type: 2)
WARN  [qmc5883] no device on bus 2 (type: 2)
WARN  [qmc5883] no device on bus 4 (type: 2)
WARN  [lis3mdl] no device on bus 2
WARN  [lis3mdl] no device on bus 2
WARN  [lis3mdl] no device on bus 2
nsh: icm20948: command not found
WARN  [ak09916] AK09916 mag init failed
ERROR [ak09916] driver start failed
IST8310 on I2C bus 3 at 0x0e (bus: 100 KHz, max: 400 KHz)
MS5611_SPI on SPI bus 4 at 0 (20000 KHz)
ERROR [battery_status] no ADC found: /dev/adc0 (2)
INFO  [commander] Mission #2 loaded, 7 WPs, curr: 0
INFO  [uavcan] Node ID 1, bitrate 1000000
INFO  [uavcan] sensor bridge 'battery' init ok
INFO  [uavcan] sensor bridge 'flow' init ok
INFO  [uavcan] sensor bridge 'gnss' init ok
INFO  [uavcan] sensor bridge 'mag' init ok
INFO  [uavcan] sensor bridge 'baro' init ok
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS0
WARN  [uavcan] GNSS ORB fd 7
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> WARN  [load_mon] wq:I2C3 low on stack! (284 bytes left)
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
`
first:UAVCAN_ENABLE = 0
second :UAVCAN_ENABLE = 1

any wrong?   

thank you!

liganggg, 2020-04-02:
[cmp1.txt](https://github.com/PX4/Firmware/files/4418523/cmp1.txt)

pavel-kirienko, 2020-04-02:
I don't see anything wrong in the log. Can you please ensure that there is no peripheral that is shared between UAVCAN and the LED in FMUv5-nano? If there doesn't seem to be any, consider stepping through UAVCAN initialization, perhaps, or maybe disable parts of the UAVCAN driver to see if it affects the LED.

julianoes, 2020-07-02:
There was no follow up, so I'm closing this for now.

closed 2020-07-02 by julianoes