PX4/PX4-Autopilot #14559: Magless operation improvements

bresch, 2020-04-01:
Magnetometer-less GPS fusion is now possible thanks to the INS/GNSS yaw estimation provided by the yaw EKF-GSF filter introduced in https://github.com/PX4/ecl/pull/770 (Firmware PR: https://github.com/PX4/Firmware/pull/14162).

The general functionality is there and works properly but a few points are still to be improved (https://github.com/PX4/Firmware/pull/14162#issuecomment-607149388):
- [ ] Set home position even if GPS is not fused in the EKF solution (because the orientation is unknown)
- [ ] Improve local position validity probation logic: we shouldn't need to wait for fore than a few seconds if the EKF reports valid local position

MaEtUgR, 2020-04-01:
The second point would require to find a good solution for https://github.com/PX4/Firmware/pull/14224

ryanjAA, 2020-04-18:
Is there anything specific or special that should be considered if wanting to test this in fixedwing?

bresch, 2020-04-19:
@ryanjAA Please try it first in SITL (I never tried it for a FW). If everything goes well, I'm quite confident that this would also work properly on a real vehicle. Just be ready to do the whole flight in stabilized/altitude mode if something goes wrong.

closed 2022-10-26 by bresch