PX4/PX4-Autopilot #14566: Get trapped into "dangerously low battery and SHUTDOWN" status

rolandash, 2020-04-01:
**Describe the bug**
Get trapped in "dangerously low battery and SHUTDOWN" status, due to missing usb connected detection in commander battery status check (>= v1.10).

Some functions has been rewrite in Commander.cpp since last year. Battery status check lines has been gathered into a separate function battery_status_check(). And shutdown_if_allowed() is added to provide some kind of another safeguard. The code looks cleaner now. however, the missing of usb power connected detection leads to trap into "dangerously low battery and SHUTDOWN" status in some cases. 
The detection is presented in previous px4 versions, but been dropped since v1.10.

This issue majorly infects MindRacer when powered by USB alone, like for calibration or debugging purpose.  In some cases it will also have impact on other FCUs. 

Details as following:
1. on booting up (powered from USB alone), MindRacer's ADC pin for battery measurement needs few seconds to reach stable. Between this booting period there is a chance the voltage on ADC pin may exceeds battery detection threshold. So even only few seconds, but already been quite enough for battery_check_status() function detects an emergency low battery and issues shutdown command. The system then will get into ARMING_STATE_SHUTDOWN status and trapped there.
2. This issue might also happen to other FCU in some cases. Considering when a FCU connected with a 3S battery, now connected to ground station through USB cable to do parameter update or calibration. If in groundstation you changes the power parameter from 3S battery to 4S battery, the system will then immediately get into ARMING_STATE_SHUTDOWN status and trapped there, unless you disconnect the battery. 

When powered from battery, MindRacer does not have this issue as PDB will smooth away all voltage fluctuations.

This issue is not observed on firmware version earlier than v1.9.x.
This issue does not exist on MindPX.

**To Reproduce**
Steps to reproduce the behavior:
1. For MindRacer users, use v1.10 or higher version of firmware.
2. connected MindRacer (standalone) to QGC through USB cable and wait for connection.
3. QGC will alert "dangerously low battery, system SHUTDOWN"
4. You can not perform most of calibration/param update actions. QGC alert "Transition denied, SHUTDOWN to INIT" or similar.

If you use other FCUs, 
1. power FCU with a 3S battery, use v1.10 or higher version of firmware.
2. connect to QGC through USB cable and wait for connection.
3. in QGC change power parameter from 3S battery to 4S battery.
4. QGC will alert "dangerously low battery, system SHUTDOWN"
5. You can not perform most of calibration/param update actions. QGC alert "Transition denied, SHUTDOWN to INIT" or similar. You need to unplug battery to get out of this.

**Expected behavior**
Not get into "dangerously low battery, system SHUTDOWN" status.

**Additional context**
This problem rather annoying when connect to ground station for calibrating/debugging.
I suggest 2 feasible fixes to this issue.
1. Add system_power.usb_connected detection in battery_status_check() function to avoid getting into shutdown status when usb cable connected.
2. Strictly the implementation of shutdown_if_allowed() is problematic. Once you get into ARMING_STATE_SHUTDOWN status you can never get out. So It is recommended to use a test-and-set 2 phase pattern instead, query if transition allowed first, so we can always rollback to a safe status if something happened in middle of transition.

Method 1 is quite straight forward and I have tested working fine on my side. Method 2 may be further improved in later releases.

@LorenzMeier @dagar Can you please help to fix this in the coming release?
thank you.

rolandash, 2020-04-02:
this may also fix #14527 @MaEtUgR

mrpollo, 2020-04-06:
@dagar @maetugr any info here? if this is something a contributor from the community can fix it would be useful to lead them to a potential fix

tacdin, 2020-04-10:
I'm having the same problem. I would be glad if you can direct the developers and users. It is very frustrating to get this error. No operations can be performed.
[image]

dagar, 2020-05-04:
Please try https://github.com/PX4/Firmware/pull/14825.

dagar, 2020-05-06:
Fixed in https://github.com/PX4/Firmware/pull/14825.

closed 2020-05-06 by dagar