PX4/PX4-Autopilot #14577: compile PX4 master on ubuntu 16.04: error: ‘Q_WS_MAC’ was not declared in this scope

FaboNo, 2020-04-02:
**Describe the bug**
I cloned the master branch, enter make px4_sitl gazebo and got the following error
excerpt from build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log :

```
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/fabrice/catkin_demo_ws/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp

Run Build Command:"/home/fabrice/ninja/ninja" "cmTC_56adf"
[1/2] Building C object CMakeFiles/cmTC_56adf.dir/CheckSymbolExists.c.o
[2/2] Linking C executable cmTC_56adf
FAILED: : && /usr/bin/cc     CMakeFiles/cmTC_56adf.dir/CheckSymbolExists.c.o  -o cmTC_56adf   && :
CMakeFiles/cmTC_56adf.dir/CheckSymbolExists.c.o : Dans la fonction « main » :
CheckSymbolExists.c:(.text+0x16) : référence indéfinie vers « pthread_create »
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

File /home/fabrice/catkin_demo_ws/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/fabrice/catkin_demo_ws/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp

Run Build Command:"/home/fabrice/ninja/ninja" "cmTC_a57ad"
[1/2] Building C object CMakeFiles/cmTC_a57ad.dir/CheckFunctionExists.c.o
[2/2] Linking C executable cmTC_a57ad
FAILED: : && /usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create   CMakeFiles/cmTC_a57ad.dir/CheckFunctionExists.c.o  -o cmTC_a57ad  -lpthreads && :
/usr/bin/ld : ne peut trouver -lpthreads
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

Determining if the Q_WS_WIN exist failed with the following output:
Change Dir: /home/fabrice/catkin_demo_ws/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp

Run Build Command:"/home/fabrice/ninja/ninja" "cmTC_bb077"
[1/2] Building CXX object CMakeFiles/cmTC_bb077.dir/CheckSymbolExists.cxx.o
FAILED: /usr/bin/c++    -I/usr/include/qt4  -o CMakeFiles/cmTC_bb077.dir/CheckSymbolExists.cxx.o -c CheckSymbolExists.cxx
CheckSymbolExists.cxx: In function ‘int main(int, char**)’:
CheckSymbolExists.cxx:8:19: error: ‘Q_WS_WIN’ was not declared in this scope
   return ((int*)(&Q_WS_WIN))[argc];
                   ^
ninja: build stopped: subcommand failed.

File /home/fabrice/catkin_demo_ws/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx:
/* */
#include <QtCore/qglobal.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef Q_WS_WIN
  return ((int*)(&Q_WS_WIN))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the Q_WS_QWS exist failed with the following output:
Change Dir: /home/fabrice/catkin_demo_ws/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp

Run Build Command:"/home/fabrice/ninja/ninja" "cmTC_358d6"
[1/2] Building CXX object CMakeFiles/cmTC_358d6.dir/CheckSymbolExists.cxx.o
FAILED: /usr/bin/c++    -I/usr/include/qt4  -o CMakeFiles/cmTC_358d6.dir/CheckSymbolExists.cxx.o -c CheckSymbolExists.cxx
CheckSymbolExists.cxx: In function ‘int main(int, char**)’:
CheckSymbolExists.cxx:8:19: error: ‘Q_WS_QWS’ was not declared in this scope
   return ((int*)(&Q_WS_QWS))[argc];
                   ^
ninja: build stopped: subcommand failed.

File /home/fabrice/catkin_demo_ws/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx:
/* */
#include <QtCore/qglobal.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef Q_WS_QWS
  return ((int*)(&Q_WS_QWS))[argc];
#else
  (void)argc;
  return 0;
#endif
}

```
Someone can tell me what is going wrong?
Thank you

julianoes, 2020-04-03:
Can you paste the full output of `make px4_sitl gazebo` after a `make clean`?

FaboNo, 2020-04-03:
@julianoes 
yes sure, here it is:
```
-- PX4 version: v1.11.0-beta1-1192-g2baa5ca
-- PX4 config file: /home/fabrice/catkin_demo_ws/src/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-248-g2fa4341
-- Configuring done
-- Generating done
-- Build files have been written to: /home/fabrice/catkin_demo_ws/src/Firmware/build/px4_sitl_default
[7/738] git submodule src/drivers/gps/devices
[13/738] git submodule src/lib/ecl
[17/738] git submodule mavlink/include/mavlink/v2.0
[18/738] git submodule Tools/sitl_gazebo
[23/738] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++14 compiler
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
[... 75 lines cut ...]
    linked by target "gazebo_usv_dynamics_plugin" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_sonar_plugin" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "mav_msgs" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "nav_msgs" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_lidar_plugin" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_plugin" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_mavlink_interface" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_uuv_plugin" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_vision_plugin" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_multirotor_base_plugin" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_imu_plugin" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gimbal_controller_plugin" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "physics_msgs" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_motor_model" in directory /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo

-- Configuring incomplete, errors occurred!
See also "/home/fabrice/catkin_demo_ws/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/fabrice/catkin_demo_ws/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[83/738] Building CXX object msg/CMakeFiles/uorb_msgs.dir/topics_sources/ekf_gps_drift.cpp.o
FAILED: cd /home/fabrice/catkin_demo_ws/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/fabrice/catkin_demo_ws/src/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/fabrice/catkin_demo_ws/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[83/738] Building CXX object msg/CMakeFiles/uorb_msgs.dir/topics_sources/ekf_gps_drift.cpp.o
ninja: build stopped: subcommand failed.
Makefile:205 : la recette pour la cible « px4_sitl » a échouée
make: *** [px4_sitl] Erreur 1
```

julianoes, 2020-04-03:
The first error I see is:
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_LIBRARIES (ADVANCED)
```
So make sure to have the gstreamer dependencies:
https://github.com/PX4/Firmware/blob/master/Tools/setup/ubuntu.sh#L192-L198

FaboNo, 2020-04-04:
@julianoes  ah yes gstreamer1.0-plugins-base was missing and now it compiles from the Firmware directory.
However, if I am doing catkin build from the catkin workspace, I am still facing the following issue:
Starting  >>> px4                                                                                                          
```
No handlers could be found for logger "trollius"ued] [hector_geotiff_plugins:make (0%) - 0.4] [mavros:make (0%) - 0.3] ... 
___________________________________________________________________________________________________________________________
Errors     << px4:check /home/fabrice/catkin_demo_ws/logs/px4/build.check.208.log                                          
CMake Error at /home/fabrice/catkin_demo_ws/src/Firmware/cmake/px4_config.cmake:81 (message):
  PX4 config file not set, try one of
  aerotenna/ocpoc/default.cmake;airmind/mindpx-v2/default.cmake;atlflight/eagle/cmake/upload.cmake;atlflight/eagle/default.cmake;atlflight/eagle/qurt.cmake;atlflight/excelsior/default.cmake;atlflight/excelsior/qurt.cmake;av/x-v1/default.cmake;beaglebone/blue/cmake/init.cmake;beaglebone/blue/cmake/upload.cmake;beaglebone/blue/default.cmake;bitcraze/crazyflie/default.cmake;emlid/navio2/cmake/upload.cmake;emlid/navio2/default.cmake;holybro/durandal-v1/bootloader.cmake;holybro/durandal-v1/default.cmake;holybro/durandal-v1/stackcheck.cmake;holybro/kakutef7/default.cmake;intel/aerofc-v1/cmake/upload.cmake;intel/aerofc-v1/default.cmake;intel/aerofc-v1/rtps.cmake;modalai/fc-v1/default.cmake;mro/ctrl-zero-f7/default.cmake;mro/x21-777/default.cmake;mro/x21/default.cmake;nxp/fmuk66-v3/default.cmake;nxp/fmurt1062-v1/default.cmake;nxp/rddrone-uavcan146/default.cmake;omnibus/f4sd/default.cmake;px4/fmu-v2/default.cmake;px4/fmu-v2/fixedwing.cmake;px4/fmu-v2/lpe.cmake;px4/fmu-v2/multicopter.cmake;px4/fmu-v2/rover.cmake;px4/fmu-v2/test.cmake;px4/fmu-v3/default.cmake;px4/fmu-v3/rtps.cmake;px4/fmu-v3/stackcheck.cmake;px4/fmu-v4/cannode.cmake;px4/fmu-v4/default.cmake;px4/fmu-v4/optimized.cmake;px4/fmu-v4/rtps.cmake;px4/fmu-v4/stackcheck.cmake;px4/fmu-v4pro/default.cmake;px4/fmu-v4pro/rtps.cmake;px4/fmu-v5/critmonitor.cmake;px4/fmu-v5/default.cmake;px4/fmu-v5/fixedwing.cmake;px4/fmu-v5/irqmonitor.cmake;px4/fmu-v5/multicopter.cmake;px4/fmu-v5/optimized.cmake;px4/fmu-v5/rover.cmake;px4/fmu-v5/rtps.cmake;px4/fmu-v5/stackcheck.cmake;px4/fmu-v5x/default.cmake;px4/fmu-v5x/p2_base_phy_LAN8742Ai.cmake;px4/io-v2/default.cmake;px4/raspberrypi/cmake/upload.cmake;px4/raspberrypi/default.cmake;px4/sitl/default.cmake;px4/sitl/replay.cmake;px4/sitl/rtps.cmake;px4/sitl/test.cmake;uvify/core/default.cmake
Call Stack (most recent call first):
  CMakeLists.txt:142 (include)

make: *** [cmake_check_build_system] Erreur 1

```
Do you have any idea about how to do?

TSC21, 2020-04-04:
> @julianoes ah yes gstreamer1.0-plugins-base was missing and now it compiles from the Firmware directory.
> However, if I am doing catkin build from the catkin workspace, I am still facing the [...]

@FaboNo what version of the Firmware you are building and what do you have on your src dir of the catkin_ws? Just to warn you that if you don't have sitl_gazebo as well at the same level of the Firmware dir in the src folder, you won't be able to use the launch files as you normally do when you build ROS packages inside a catkin_ws and after sourcing devel/setup.bash, unless you manually set paths to the Gazebo resource and model paths.

You can have a look at the CI stage where we test this: https://github.com/PX4/Firmware/blob/master/Jenkinsfile#L23. Note that although we support this (in a limited way), this is not a conventional way of building the Firmware.

FaboNo, 2020-04-04:
@TSC21  
I am building master because I would like to test the rover controller and from my undrstanding a running version is available in master.
Under src, I have mavros, Firmware, mavlink, mavcom and other packages.
sitl_gazebo is built in src/Firmware/Tools as "usual" I will say. 

So if I understand correctly, sitl_gazebo should be located in src and not in src/Firmware/Tools?

TSC21, 2020-04-04:
> @TSC21
> I am building master because I would like to test the rover controller and from my undrstanding a running version is available in master.
> Under src, I have mavros, Firmware, mavlink, mavcom [...]

You can leave it in Tools as well, and create a symlink to it from the catkin_ws. "Usual" doesn't mean anything if the build setup by itself is completely different. As I said, using catkin is supported but quite limited.

TSC21, 2020-04-04:
Also if you want to test the rover you don't need to build the Firmware in a catkin_ws. It's not even the recommended setup considering that we don't fully support catkin (example: the output dirs are not compatible with the ones expected from a ROS package standpoint)

FaboNo, 2020-04-05:
@TSC21 in Firmware there are launch files such as mavros_posix_sitl.launch.  It means that i should be able to run catkin build so that the px4 packahe is recognized as well as the launch files. 
but with master, I cannot make it! I still get the error I mentioned before:
```
Errors     << px4:check /home/fabrice/catkin_demo_ws/logs/px4/build.check.210.log                                                
CMake Error at /home/fabrice/catkin_demo_ws/src/Firmware/cmake/px4_config.cmake:81 (message):
  PX4 config file not set, try one of
  aerotenna/ocpoc/default.cmake;airmind/mindpx-v2/default.cmake;atlflight/eagle/cmake/upload.cmake;atlflight/eagle/default.cmake;atlflight/eagle/qurt.cmake;atlflight/excelsior/default.cmake;atlflight/excelsior/qurt.cmake;av/x-v1/default.cmake;beaglebone/blue/cmake/init.cmake;beaglebone/blue/cmake/upload.cmake;beaglebone/blue/default.cmake;bitcraze/crazyflie/default.cmake;emlid/navio2/cmake/upload.cmake;emlid/navio2/default.cmake;holybro/durandal-v1/bootloader.cmake;holybro/durandal-v1/default.cmake;holybro/durandal-v1/stackcheck.cmake;holybro/kakutef7/default.cmake;intel/aerofc-v1/cmake/upload.cmake;intel/aerofc-v1/default.cmake;intel/aerofc-v1/rtps.cmake;modalai/fc-v1/default.cmake;mro/ctrl-zero-f7/default.cmake;mro/x21-777/default.cmake;mro/x21/default.cmake;nxp/fmuk66-v3/default.cmake;nxp/fmurt1062-v1/default.cmake;nxp/rddrone-uavcan146/default.cmake;omnibus/f4sd/default.cmake;px4/fmu-v2/default.cmake;px4/fmu-v2/fixedwing.cmake;px4/fmu-v2/lpe.cmake;px4/fmu-v2/multicopter.cmake;px4/fmu-v2/rover.cmake;px4/fmu-v2/test.cmake;px4/fmu-v3/default.cmake;px4/fmu-v3/rtps.cmake;px4/fmu-v3/stackcheck.cmake;px4/fmu-v4/cannode.cmake;px4/fmu-v4/default.cmake;px4/fmu-v4/optimized.cmake;px4/fmu-v4/rtps.cmake;px4/fmu-v4/stackcheck.cmake;px4/fmu-v4pro/default.cmake;px4/fmu-v4pro/rtps.cmake;px4/fmu-v5/critmonitor.cmake;px4/fmu-v5/default.cmake;px4/fmu-v5/fixedwing.cmake;px4/fmu-v5/irqmonitor.cmake;px4/fmu-v5/multicopter.cmake;px4/fmu-v5/optimized.cmake;px4/fmu-v5/rover.cmake;px4/fmu-v5/rtps.cmake;px4/fmu-v5/stackcheck.cmake;px4/fmu-v5x/default.cmake;px4/fmu-v5x/p2_base_phy_LAN8742Ai.cmake;px4/io-v2/default.cmake;px4/raspberrypi/cmake/upload.cmake;px4/raspberrypi/default.cmake;px4/sitl/default.cmake;px4/sitl/replay.cmake;px4/sitl/rtps.cmake;px4/sitl/test.cmake;uvify/core/default.cmake
Call Stack (most recent call first):
  CMakeLists.txt:142 (include)

make: *** [cmake_check_build_system] Erreur 1
cd /home/fabrice/catkin_demo_ws/build/px4; catkin build --get-env px4 | catkin env -si  /usr/bin/make cmake_check_build_system; cd -

```
However, if as you said, it is not necessary to build px4 under a catkin workspace to test the rover, what do you suggest (as I will need mavros as well)?

TSC21, 2020-04-05:
> @TSC21 in Firmware there are launch files such as mavros_posix_sitl.launch. It means that i should be able to run catkin build so that the px4 packahe is recognized as well as the launch files.

I know that but if you understand what I said before, you will see that these launch files are not installed in a way that they can be used.

> However, if as you said, it is not necessary to build px4 under a catkin workspace to test the rover, what do you suggest (as I will need mavros as well)?

Lots of projects build and run PX4 SITL together with ROS without having the project build under a catkin ws. Just build PX4 SITL the default way and then run MAVROS linked with the UDP port of PX4 SITL. If you take a look at the launch files in PX4, you can easily understand what needs to be done.

TSC21, 2020-04-05:
@FaboNo a good example of a ROS project that uses PX4 SITL without building it inside a catkin ws is PX4 Avoidance - https://github.com/PX4/avoidance#run-the-avoidance-gazebo-simulation.

TSC21, 2020-04-05:
I am closing this issue since the initial problem was solved and it diverged to a different subject. Please bring your questions into a topic in discuss.px4.io.

closed 2020-04-05 by TSC21