PX4/PX4-Autopilot #14578: board support i2c improvements

dagar, 2020-04-02:
The recent i2c/spi driver changes and corresponding board level support has been a solid step forward, but one small thing we've lost is the configuration of on board I2C devices. In many cases the I2C address is a board specific configuration that shouldn't be pushed out to the driver.

Similar to px4_spi_buses (hierarchical configuration of SPI devices per bus) should we add something similar for on board I2C?

 - I2C1 (internal/external)
    - ist8310: 4 possible i2c addresses based on how 2 pins are connected. Max speed 400 kHz 
    - rgbled_ncp5623c: i2c address is fixed. Max speed 400 kHz 

This could also be used to set the bus frequency appropriately, rather than defaulting to 100 kHz in most cases.

dagar, 2020-04-02:
@davids5 any thing you'd like to see here?

davids5, 2020-04-02:
>Similar to px4_spi_buses (hierarchical configuration of SPI devices per bus) should we add something similar for on board I2C?

At first blush we would assume the bus carries the external/external but I think we need to add it to the device.
Yes bus, enternal/external;, address, max clock, and pehaps max message size

max clock - min of all the max clock is the bus freq used.
max message size could be used for bandwidth calkc at the actual bus freq used

davids5, 2020-04-02:
and maybe drdry?

dagar, 2020-04-02:
> At first blush we would assume the bus carries the external/external but I think we need to add it to the device.

I'm not sure what you mean, can you expand on this?

dagar, 2020-04-02:
> and maybe drdry?

Yes, at the moment we have `spi_drdy_gpio_t`, but there are a few cases where we might as well use them for an I2C device as well.

davids5, 2020-04-02:
> > At first blush we would assume the bus carries the internal/external but I think we need to add it to the device.
> 
> I'm not sure what you mean, can you expand on this?

Rephrased: At first blush we would assume the bus carries the internal/external attribute.

I have seen HW were a bus goes to device on board (Address A) and on to a connector connected to  same device (Address B) that is off board that forms the system. 

So even if the connector exist for a bus it does not mean say a mag on that bus is external.

dagar, 2020-04-02:
> I have seen HW were a bus goes to device on board (Address A) and on to a connector connected to same device (Address B) that is off board that forms the system.

Ok, so a few come to mind like the nxphlite, omnibus, kakute, and probably others. I think we'd actually be in a good position to handle this as is with the external() check on the interface (I2C class). At the board level we could have internal/external/mixed bus. In the mixed case the internal sensors would be marked and I2C interface external() could check the board's special cases.

bkueng, 2020-04-03:
Yes it's something I considered adding as well. That way we we also know which devices have to be present.

> At the board level we could have internal/external/mixed bus. In the mixed case the internal sensors would be marked and I2C interface external() could check the board's special cases.

Makes sense.

dagar, 2020-04-03:
> That way we we also know which devices have to be present.

Yes, that's exactly what I want to be able to do. 

Did you consider carrying any additional board specific sensor configuration in px4_spi_buses like the rotation or max speed?

bkueng, 2020-04-06:
> Did you consider carrying any additional board specific sensor configuration in px4_spi_buses like the rotation or max speed?

Yes, but I went for what we already have via CLI. Either way works for me, just that CLI allows for easier testing and having a longer list of optional arguments does not work well in C++ if you only want to fill in some (C++20 improves this with designated initializers). Using objects with in the form `SPIDevice(DRV_xy).maxFrequency(1000).rotation(YAW90).DRDY(xy)` is an alternative that should work as well though.

dagar, 2020-07-02:
Slight hijack, but another one that comes up occasionally is when a sensor actually has a reset line.