PX4/PX4-Autopilot #14579: [uorb] sensor_mag advertise failed (5)

Thijs----, 2020-04-02:
**Describe the bug**
On the Pixhawk cube, the ULog file is full off these logged messages: "[uorb] sensor_mag advertise failed (5)"

**To Reproduce**
Steps to reproduce the behavior:
1. Upload the latest stable PX4 firmware (v1.10.1)
2. Boot UAV and start logging (via shell "logger on")
3. Stop logging after a short time ("logger stop")
4. Open ULog, see messages

**Expected behavior**
There should be no error when the sensor_mag message is advertised

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Uploaded fresh firmware, no configuration done ([PX4 Flight Review](https://logs.px4.io/plot_app?log=2ddaa946-d048-4160-a762-a151b25efb83)).
- Did the basic config (choose airframe, calibrate sensors, etc) [PX4 Flight Review](https://logs.px4.io/plot_app?log=ceb8e90c-d4f8-4a30-8de9-fe4f771f944b).

**Drone (please complete the following information):**
- Octocopter
- PX4 Cube

**Additional context**
This started when uploaded a modified 1.10.0 version for the first time, previously there was a modified 1.8.x version of the PX4 firmware. 
On a second Pixhawk (Pixhawk 1) this problem does not occur.

dagar, 2020-04-02:
At the moment the system only supports 4 mags, although it could be increased if actually needed.

In your log I see 5.
 - mpu9250 x 2 (internal)
 - lsm303d (internal)
 - hmc5883 (external mag)
 - ak09916 (external mag)

In the future we're considering not even enabling the mpu9250 mags by default. They don't work that well internally and having the mag enabled messes with the sampling rate of the IMU.

Thijs----, 2020-04-03:
Hi Dagar,

Thanks for the fast and clear answer! 

I use the Pixhawk Cube and the here+ gps.
Today I'll check if all these sensors are correctly detected, because did not expect this many mags from only those two devices. If they are correct, I'll disable the mpu9250 mags thanks to your advice.

I'll report my findings here, and will then close the issue.

Thijs----, 2020-04-03:
Ok, indeed the cube has 2x MPU9250 and 1x LSM303D
I cannot really confirm that hmc5883 and ak09916 are in the here+, but it is possible.

Now I am a bit confused how I should disable both internal mpu9250 and use the three other mags. @dagar , can you point me in the correct direction?

I already thought of two possibilities:
1. Use the CAL_MAGX_EN, but how can I the MAGX_ID of each mag? + If I disable the 2 correct internal mag, how can I enable (and calibrate) the 5th mag?
2. Change the mpu9250 driver so the mags are not used, but this seems to throw other issues.

So my possibilities does not seem the be the way to do it.

Thijs----, 2020-04-03:
I see two approaches:
1. disable the start of the two MPU9250 sensor, so in total the system only sees 3 mags = easiest way. Can maybe be achieved by modifying https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v3/init/rc.board_sensors
2. allow 5 mags, but I think a lot need to change: sensor_mag topic, calibration, ...

Can you help me with one of the approaches, or suggest a better solution?

closed 2020-04-14 by Thijs----