PX4/PX4-Autopilot #14580: Validate builds over GNU ARM GCC Toolchain 9

TSC21, 2020-04-02:
Following https://github.com/PX4/Firmware/pull/14159, we need to validate the generated binaries in some flights. We can rather provide the binaries or provide instructions on how to install the new toolchain so they can be generated locally,

@PX4/testflights what is the best way to proceed here?

Thank you!

@MaEtUgR @dagar @LorenzMeier FYI

dannyfpv, 2020-04-06:
tested on pixhawk4 v5 f-450

**Modes Tested**

Position Mode: Good.
Altitude Mode: Good.
Stabilized Mode: Good.
Mission Plan Mode (Automated): Good.
RTL: Good.

**- Procedure**
Arm and Take off in position mode, after flying for approximately one minute, switched to altitude then stabilized mode proceed to switch to mission plan mode then make sure that vehicle follows all waypoints as shown in QGC, once completed all waypoin activate RTL. 

**Notes:**
No issues noted, good flight in general.

**Log:**

indoor flight log:
https://review.px4.io/plot_app?log=261e38b1-d7ac-4570-95fa-99594a160ad3

mission mode log:
https://review.px4.io/plot_app?log=23faea6e-298e-48ce-ad8d-83ec575a0d9d

MaEtUgR, 2020-04-07:
I sent @Tony3dr GCC 9 built binaries for the other targets he requested.
I flew an fmu-v5 plane and an omnibus racer myself already and had no related issues so far.

jorge789, 2020-04-08:
### Tested on PixRacer V4
**Indoor Flight Test**
Modes Tested
Stabilized Mode: Good.

**Procedure**
Arm and Take off in stabilized mode.

**Notes**
No issues noted, good flight in general.

**Log**
https://review.px4.io/plot_app?log=3959d2ea-60c4-42c2-8a73-117c40cb389b

https://review.px4.io/plot_app?log=11f463c1-d600-4117-bc77-7f639c67cef6

Junkim3DR, 2020-04-08:
### Tested on Pixhawk3 V4Pro

**Indoor test flight.**
Mode tested in Stabilized

**Procedure**
Takeoff and landing in Stabilized

**Note**
No issue noted.

**Log**
- https://review.px4.io/plot_app?log=94ad8fdd-b548-4ebf-8c57-36efc982d467
- https://review.px4.io/plot_app?log=1f3b9df5-58a6-4489-ab67-e0d59b0c5c2b

TSC21, 2020-04-08:
@PX4/testflights excellent! Thank you very much!

TSC21, 2020-04-09:
@dagar @MaEtUgR should we consider this validated?

MaEtUgR, 2020-04-15:
> should we consider this validated?

To not cause significant issues immediately yes. Since it's supported on master I'm doing all my development tests using GCC 9 and didn't find a problem yet.