PX4/PX4-Autopilot #14588: [Gazebo][Tailsitter] Drone angle is unstable when fw to mc

xdwgood, 2020-04-03:
**Describe the bug**
I used the latest mastre branch and performed the gazebo_tailsitter test. I found that when I choose fw to mc, the drone attitude is unstable!

I'm not sure if this is a simulation problem? Or is there a mistake in the recent changes?

**To Reproduce**

1. make px4_sitl gazebo_tailsitter
2. open QGC ,send takeoff
3. switch mc to fw
4. Wait a minute
5 switch fw to mc
6. see unstable angle

**Expected behavior**
have stable back-transmission

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=e5626890-f8fe-4432-b463-2ab4daa8b579

xdwgood, 2020-04-09:
@sfuhrer could you have a look？ thanks

xdwgood, 2020-04-13:
I tried v1.10 and everything works fine in gazebo

sfuhrer, 2020-04-14:
@xdwgood good to know that the problem isn't there on the release. It could be that there was some underlying problem in gazebo that was fixed since. 
Do you still want to keep this issue open? Have you also done real flight testing with the release?

xdwgood, 2020-04-14:
I plan to make a flight to the master branch, but I found this in the simulation, I am worried about the safety of outdoor flight.
So am I trying to understand that this is just a simulation problem?....But I'm not sure yet. I tried v1.11.0-BETA1 and had the same problem ...
@sfuhrer  Could you see the same problem with the gazebo simulation on your computer?

closed 2020-04-30 by xdwgood

sfuhrer, 2020-05-02:
@xdwgood sorry for the late reply, this kind of slipped. 
I just also tried in SITL, didn't have issues (beside a very jumpy airspeed reading). Maybe that was causing issues in your case. 
Did you also fly with a real tailsitter lately?

xdwgood, 2020-05-09:
@sfuhrer   

>  Did you also fly with a real tailsitter lately?

yes, I tried a real flight and it worked well.