PX4/PX4-Autopilot #14593: Adding support for Crazyflie 2.1

shrit, 2020-04-04:
Hello

I have tried to flash PX4 on crazyflie 2.1 using Qgroundcontrol. 
After installing the new firmware (stable, beta, and the master). Qground control showed a message `Parameters are missing from firmware` with a list of parameters
The vehicle does not seem to answer any command sent by Qground control such as (arm, disarm)
I was not able to calibrate sensors, or turning the motors.
Documentation only shows support for crazyflie 2.0 but not with crazyflie 2.1. Since the differences between these two are minor (a new IMU and a new Barometers), Is it possible to have 2.1 supported?

**To Reproduce**
Try to flash PX4 on crazyflie 2.1

**Log Files and Screenshots**
Here is the log of Qground control saved as txt [log.txt](https://github.com/PX4/Firmware/files/4431786/log.txt)

Best regards

TSC21, 2020-04-04:
@shrit The community is free to provide support to these boards if they are willing to. So that is what I recommend you try to do.

dagar, 2020-07-24:
Closing as duplicate of https://github.com/PX4/Firmware/issues/15361.

closed 2020-07-24 by dagar