PX4/PX4-Autopilot #14600: logger watchdog triggered on FW fmu-v5

dagar, 2020-04-05:
https://logs.px4.io/plot_app?log=b815c6b1-707c-4a9b-bdf3-5bb7f41af758
CPU usage went to 100% shortly after arming due to the logger watchdog triggering.

dagar, 2020-04-05:
Unfortunately no post flight top or perf counter output.

dagar, 2020-04-05:
There's top output captured when watchdog triggered, but it's not displayed on flight review (we should add it).

``` Console
PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
442 mavlink_rcv_if0             89477 45.854  2600/ 3908 175 (175)  READY  4
```

Something went wrong in the first mavlink instance receive thread.

dagar, 2020-04-05:
Based on the mavlink configuration (only MAV_0_CONFIG set) I believe that first mavlink instance is USB.

```
MAV_0_CONFIG, 101
MAV_1_CONFIG, 0
MAV_2_CONFIG, 0
```

dagar, 2020-04-05:
@MaEtUgR can you confirm if USB was connected or not?
[image]

dagar, 2020-04-05:
Note for later, re-evaluate `commander` priority. We can structure it such that things like mavlink, logging, gimbal, etc is unable to interfere with the safety critical state machine operation.

[image]

MaEtUgR, 2020-04-05:
So what I did is try to arm by RC after powering and waiting it didn't even after quite some tries. Then I opened the lid (containing GPS and external mag) and plugged the USB cable which is attached to the JST-GH with a female USB-A to male USB-C to my phone and see what happened in QGC. I tried to arm again and it armed. So I disarmed, unplugged USB, put the lid back and tried to arm without success. So I repeated the same procedure again except I didn't disarm in between but just kept the throttle on zero (auto disarm is disabled). I realized stabilized mode doesn't work so I checked my surfaces manual and after verifying them I started flying manual and tried to switch to stabilized once in air but nothing happened.

Note: The exact same setup at home on the floor one hour before did arm and switch to stabilized and I didn't attach any USB.

MaEtUgR, 2020-04-05:
When I try to arm it on the bench now it complains about inconsistent compasses and when I force arm it stays within normal CPU load levels <65%. The compass inconsistency is an old problem on this one, I recalibrated them over and over again (with the lid on of course) and they are inconsistent again. I'll probably just disable the internal one to get rid of the error.

dagar, 2020-04-05:
This is a bit of a leap, but so far the removal of the mavlink receiver sleep comes to mind (https://github.com/PX4/Firmware/pull/13764). Something to be aware of when considering possible NuttX USB CDC ACM related issues.

MaEtUgR, 2020-04-05:
I managed to reproduce indoors using the battery: https://logs.px4.io/plot_app?log=0e19ad39-ed6d-4680-95b1-e74184f7cf74

1. I plugged the battery and tried to arm but it didn't, red flashing LED
2. I opened the lid and plugged USB to my phone just like on the field, this time it immediately showed me the messages from the tries before and showed compass inconsistency
3. I tried to arm but it said compass still inconsistent
4. I moved the lid out of the right orientation until it was "consistent" with the internal one
5. I got accel bias to high
6. I rebooted and tried to arm again and it did
7. I tried switching to stabilized and it worked, see `nav_state` in the log
8. I unplugged my phone and wasn't able to switch mode anymore or disarm

It could be unplugging USB while armed or QGC on the phone is still connected.

MaEtUgR, 2020-04-05:
Doesn't it try to reboot when I unplug USB but when it's armed it's not allowed so 💥 ?

PR #14602 (2020-04-06): mavlink: fix potential busy loop upon USB disconnect
When disconnecting USB poll returns successfully, but the read returns
-1 with ENOTCONN.
In addition this also ensures there's no busy loop when poll returns an
error.

MAVLink continues to work after reconnecting USB.

fixes  #14600

@MaEtUgR
closed 2020-04-06 by dagar