PX4/PX4-Autopilot #14601: FAILED: uORB

pablomarcel, 2020-04-05:
**Describe the bug**
Tried to run a simulation using this: make px4_sitl_default jmavsim
and got the errors posted below

**To Reproduce**
Steps to reproduce the behavior:
Using macOS Catalina
1. in terminal: px4_sitl_default jmavsim
2. gets to this line: Build files have been written to: /Users/...
3. then breaks: FAILED: uORB/topics/uORBTopics.hpp...
4. Failed to import em: No module named 'em'...
5. You may need to install it using: pip3 install --user empy

which makes no sense

**Expected behavior**
just run the program without a hassle

**Additional context**
"
[1/728] Generating uORB topic headers
FAILED: uORB/topics/uORBTopics.hpp uORB/topics/actuator_armed.h uORB/topics/actuator_controls.h uORB/topics/actuator_outputs.h uORB/topics/adc_report.h uORB/topics/airspeed.h uORB/topics/airspeed_validated.h uORB/topics/battery_status.h uORB/topics/camera_capture.h uORB/topics/camera_trigger.h uORB/topics/cellular_status.h uORB/topics/collision_constraints.h uORB/topics/collision_report.h uORB/topics/commander_state.h uORB/topics/cpuload.h uORB/topics/debug_array.h uORB/topics/debug_key_value.h uORB/topics/debug_value.h uORB/topics/debug_vect.h uORB/topics/differential_pressure.h uORB/topics/distance_sensor.h uORB/topics/ekf2_timestamps.h uORB/topics/ekf_gps_drift.h uORB/topics/ekf_gps_position.h uORB/topics/esc_report.h uORB/topics/esc_status.h uORB/topics/estimator_innovations.h uORB/topics/estimator_sensor_bias.h uORB/topics/estimator_status.h uORB/topics/follow_target.h uORB/topics/geofence_result.h uORB/topics/gps_dump.h uORB/topics/gps_inject_data.h uORB/topics/home_position.h uORB/topics/hover_thrust_estimate.h uORB/topics/input_rc.h uORB/topics/iridiumsbd_status.h uORB/topics/irlock_report.h uORB/topics/landing_gear.h uORB/topics/landing_target_innovations.h uORB/topics/landing_target_pose.h uORB/topics/led_control.h uORB/topics/log_message.h uORB/topics/logger_status.h uORB/topics/manual_control_setpoint.h uORB/topics/mavlink_log.h uORB/topics/mission.h uORB/topics/mission_result.h uORB/topics/mount_orientation.h uORB/topics/multirotor_motor_limits.h uORB/topics/obstacle_distance.h uORB/topics/offboard_control_mode.h uORB/topics/onboard_computer_status.h uORB/topics/optical_flow.h uORB/topics/orb_test.h uORB/topics/orb_test_large.h uORB/topics/orb_test_medium.h uORB/topics/orbit_status.h uORB/topics/parameter_update.h uORB/topics/ping.h uORB/topics/position_controller_landing_status.h uORB/topics/position_controller_status.h uORB/topics/position_setpoint.h uORB/topics/position_setpoint_triplet.h uORB/topics/power_button_state.h uORB/topics/power_monitor.h uORB/topics/pwm_input.h uORB/topics/qshell_req.h uORB/topics/qshell_retval.h uORB/topics/radio_status.h uORB/topics/rate_ctrl_status.h uORB/topics/rc_channels.h uORB/topics/rc_parameter_map.h uORB/topics/rpm.h uORB/topics/safety.h uORB/topics/satellite_info.h uORB/topics/sensor_accel.h uORB/topics/sensor_accel_fifo.h uORB/topics/sensor_accel_integrated.h uORB/topics/sensor_accel_status.h uORB/topics/sensor_baro.h uORB/topics/sensor_combined.h uORB/topics/sensor_correction.h uORB/topics/sensor_gyro.h uORB/topics/sensor_gyro_fifo.h uORB/topics/sensor_gyro_integrated.h uORB/topics/sensor_gyro_status.h uORB/topics/sensor_mag.h uORB/topics/sensor_preflight.h uORB/topics/sensor_selection.h uORB/topics/servorail_status.h uORB/topics/subsystem_info.h uORB/topics/system_power.h uORB/topics/task_stack_info.h uORB/topics/tecs_status.h uORB/topics/telemetry_status.h uORB/topics/test_motor.h uORB/topics/timesync.h uORB/topics/timesync_status.h uORB/topics/trajectory_bezier.h uORB/topics/trajectory_waypoint.h uORB/topics/transponder_report.h uORB/topics/tune_control.h uORB/topics/uavcan_parameter_request.h uORB/topics/uavcan_parameter_value.h uORB/topics/ulog_stream.h uORB/topics/ulog_stream_ack.h uORB/topics/vehicle_acceleration.h uORB/topics/vehicle_air_data.h uORB/topics/vehicle_angular_acceleration.h uORB/topics/vehicle_angular_velocity.h uORB/topics/vehicle_attitude.h uORB/topics/vehicle_attitude_setpoint.h uORB/topics/vehicle_command.h uORB/topics/vehicle_command_ack.h uORB/topics/vehicle_constraints.h uORB/topics/vehicle_control_mode.h uORB/topics/vehicle_global_position.h uORB/topics/vehicle_gps_position.h uORB/topics/vehicle_imu.h uORB/topics/vehicle_land_detected.h uORB/topics/vehicle_local_position.h uORB/topics/vehicle_local_position_setpoint.h uORB/topics/vehicle_magnetometer.h uORB/topics/vehicle_odometry.h uORB/topics/vehicle_rates_setpoint.h uORB/topics/vehicle_roi.h uORB/topics/vehicle_status.h uORB/topics/vehicle_status_flags.h uORB/topics/vehicle_trajectory_bezier.h uORB/topics/vehicle_trajectory_waypoint.h uORB/topics/vtol_vehicle_status.h uORB/topics/wheel_encoders.h uORB/topics/wind_estimate.h uORB/topics/yaw_estimator_status.h 
cd /Users/pablomontijo/Firmware/msg && /usr/local/bin/python3 tools/px_generate_uorb_topic_files.py --headers -f actuator_armed.msg actuator_controls.msg actuator_outputs.msg adc_report.msg airspeed.msg airspeed_validated.msg battery_status.msg camera_capture.msg camera_trigger.msg cellular_status.msg collision_constraints.msg collision_report.msg commander_state.msg cpuload.msg debug_array.msg debug_key_value.msg debug_value.msg debug_vect.msg differential_pressure.msg distance_sensor.msg ekf2_timestamps.msg ekf_gps_drift.msg ekf_gps_position.msg esc_report.msg esc_status.msg estimator_innovations.msg estimator_sensor_bias.msg estimator_status.msg follow_target.msg geofence_result.msg gps_dump.msg gps_inject_data.msg home_position.msg hover_thrust_estimate.msg input_rc.msg iridiumsbd_status.msg irlock_report.msg landing_gear.msg landing_target_innovations.msg landing_target_pose.msg led_control.msg log_message.msg logger_status.msg manual_control_setpoint.msg mavlink_log.msg mission.msg mission_result.msg mount_orientation.msg multirotor_motor_limits.msg obstacle_distance.msg offboard_control_mode.msg onboard_computer_status.msg optical_flow.msg orb_test.msg orb_test_large.msg orb_test_medium.msg orbit_status.msg parameter_update.msg ping.msg position_controller_landing_status.msg position_controller_status.msg position_setpoint.msg position_setpoint_triplet.msg power_button_state.msg power_monitor.msg pwm_input.msg qshell_req.msg qshell_retval.msg radio_status.msg rate_ctrl_status.msg rc_channels.msg rc_parameter_map.msg rpm.msg safety.msg satellite_info.msg sensor_accel.msg sensor_accel_fifo.msg sensor_accel_integrated.msg sensor_accel_status.msg sensor_baro.msg sensor_combined.msg sensor_correction.msg sensor_gyro.msg sensor_gyro_fifo.msg sensor_gyro_integrated.msg sensor_gyro_status.msg sensor_mag.msg sensor_preflight.msg sensor_selection.msg servorail_status.msg subsystem_info.msg system_power.msg task_stack_info.msg tecs_status.msg telemetry_status.msg test_motor.msg timesync.msg timesync_status.msg trajectory_bezier.msg trajectory_waypoint.msg transponder_report.msg tune_control.msg uavcan_parameter_request.msg uavcan_parameter_value.msg ulog_stream.msg ulog_stream_ack.msg vehicle_acceleration.msg vehicle_air_data.msg vehicle_angular_acceleration.msg vehicle_angular_velocity.msg vehicle_attitude.msg vehicle_attitude_setpoint.msg vehicle_command.msg vehicle_command_ack.msg vehicle_constraints.msg vehicle_control_mode.msg vehicle_global_position.msg vehicle_gps_position.msg vehicle_imu.msg vehicle_land_detected.msg vehicle_local_position.msg vehicle_local_position_setpoint.msg vehicle_magnetometer.msg vehicle_odometry.msg vehicle_rates_setpoint.msg vehicle_roi.msg vehicle_status.msg vehicle_status_flags.msg vehicle_trajectory_bezier.msg vehicle_trajectory_waypoint.msg vtol_vehicle_status.msg wheel_encoders.msg wind_estimate.msg yaw_estimator_status.msg -i /Users/pablomontijo/Firmware/msg -o /Users/pablomontijo/Firmware/build/px4_sitl_default/uORB/topics -e templates/uorb -t /Users/pablomontijo/Firmware/build/px4_sitl_default/msg/tmp/headers -q
Failed to import em: No module named 'em'

You may need to install it using:
    pip3 install --user empy

[6/728] Building CXX object platforms/posix/src/px4/common/lockstep_scheduler/CMakeFiles/lockstep_scheduler.dir/src/lockstep_scheduler.cpp.o
ninja: build stopped: subcommand failed.
make: *** [px4_sitl_default] Error 1
"

TSC21, 2020-04-05:
What doesn't make no sense? It is quite clear what is missing and it even provides you a command to install it (`pip3 install --user empy`).

pablomarcel, 2020-04-05:
it is already installed

TSC21, 2020-04-05:
There are two possibilities: 1. You have it installed under Python 3 but the build is finding Python 2; 2. You have it installed under Python 2 and the build requires Python 3.
So make sure to run that script again, that `which python` points to Python 3 and that if you grep `em` (or `empy`) you find it installed under the Python3 lib path

TSC21, 2020-04-05:
There's a big probability that you have a mix between Python libs installed by pip and the ones installed by homebrew.

pablomarcel, 2020-04-06:
noted

julianoes, 2020-04-08:
Please paste the output of:
```
which python
which python3
which pip
which pip3
python --version
python3 --version
pip --version
pip3 --version
```

DavidMM-sudomake, 2020-05-21:
Hello, I am having the exact same issue I think. The outputs for the previous commands are:

$ which python
/usr/bin/python
$ which python3
/usr/bin/python3
$ which pip
/usr/local/bin/pip
$ which pip3
_(no output)_
$ python --version
Python 2.7.17
$ python3 --version
Python 3.6.9
$ pip --version
pip 20.1.1 from /usr/local/lib/python2.7/dist-packages/pip (python 2.7)
$ pip3 --version

No se ha encontrado la orden «pip3», pero se puede instalar con:

sudo apt install python3-pip

----------------
Moreover, after running the last command, and theoretically installing python3-pip, if I introduce the previous set of commands again, the output is still the same as the one pasted.

Thanks in advance, hope you can help me, as I am having lots of problems...

julianoes, 2020-05-25:
The problem is that you have python-pip installed separately from your system. I suggest to remove (or move to backup location) `/usr/local/bin/pip`.

And then make sure to do `sudo apt install python3-pip`.

And then make sure the dependencies are installed:
```
pip3 install --user empy
```

DavidMM-sudomake, 2020-05-31:
I dont understand what you mean. I am trying to move the "pip" folder from usr/local/lib/python2.7/dist-packages/ to usr/local/bin, but it says i am not allowed to do that movement. 

When running the commands:

sudo apt install python3-pip

Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
python3-pip ya está en su versión más reciente (9.0.1-2.3~ubuntu1.18.04.1).
Los paquetes indicados a continuación se instalaron de forma automática y ya no son necesarios.
  efibootmgr gir1.2-geocodeglib-1.0 libfwup1 libllvm8 libmbim-glib4
  libmbim-proxy libqmi-glib5 libqmi-proxy python-bs4 python-html5lib
  python-lxml python-webencodings ubuntu-web-launchers usb-modeswitch
  usb-modeswitch-data
Utilice «sudo apt autoremove» para eliminarlos.
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 6 no actualizados.

pip3 install --user empy

No se ha encontrado la orden «pip3», pero se puede instalar con:

sudo apt install python3-pip

------------------
Therefore, somehow after installing python3 and after entering the dependencies command it tells me to install python3...

julianoes, 2020-06-02:
Try:
```
sudo apt-get install --reinstall python3-pip
```

julianoes, 2020-09-01:
There was no answer, closing.

closed 2020-09-01 by julianoes