PX4/PX4-Autopilot #14612: imu rotation issue

NicolasM0, 2020-04-07:
**Describe the bug**
The first IMU of my cube black have a bad rotation.

**To Reproduce**
- flash a cube black with master
- (reset all acc off to 0 and scale to 1 and reboot)
- see estimated orientation or sensors measures (my cube was leveled) :
[image]

**Expected behavior**
Acc 0 should measure about -9.81m/s/s not +9.81

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=9531c76c-abd9-4656-9fd1-3d50dbc8ced5

**Additional context**
@bkueng it was ok at 2f3174abb0ee2503a94d8b41b99e729fc1946ebb and not after 532ccd18ad2d82f72a33d1c0915a50b49f53eada

Another issue happened when I tried to calibrated the acceleros with this issue: scales of imu 1 and 2 were negative.
Shouldn't be great to have test on this

dagar, 2020-04-07:
I'm not seeing this on current master (https://github.com/PX4/Firmware/commit/4e0441ab0b9487d3bcd87cdc3373575c0fb1c9f1) or master just before we switched to the new Invensense IMU drivers (https://github.com/PX4/Firmware/commit/7b12a215650935c144c7b58ad731172a17c897d9).

[image]

[image]

Is this a new cube?

NicolasM0, 2020-04-07:
ok works now on 4e0441ab0b9487d3bcd87cdc3373575c0fb1c9f1
thank you !
I retried on 4cdc58ce8da43d2b193719982be091929267b4d0 were I was before and still have the issue.
The few commits between these should have fixed that

NicolasM0, 2020-04-07:
no not a new cube

closed 2020-04-07 by NicolasM0

NicolasM0, 2020-04-08:
@dagar I am not sure of what you mean by a "new cube"

dagar, 2020-04-08:
> @dagar I am not sure of what you mean by a "new cube"

There's apparently a new version of the Cube black (Cube Black+ ?) that changed the sensors (mpu9250 is end of life), but I don't know the details.

NicolasM0, 2020-04-08:
> > @dagar I am not sure of what you mean by a "new cube"
> 
> There's apparently a new version of the Cube black (Cube Black+ ?) that changed the sensors (mpu9250 is end of life), but I don't know the details.

I think this cube is about 1 year old