PX4/PX4-Autopilot #14623: logger profile for full raw IMU logging

dagar, 2020-04-08:
We should create a logger profile (or mode?) for logging all raw data from the new IMU drivers.

 - log a particular `sensor_accel_fifo` or `sensor_gyro_fifo` message at max rate
 - can logger keep up with a 1 or 2 kHz orb publication or do we need to queue?
 - do we need to optimize `sensor_accel_fifo` and `sensor_gyro_fifo`?
      - contains 32 element arrays of 16 bit data, but often only 4-16 samples are populated
      - if necessary for logging we could merge this data into another topic with all 32 samples populated
      - another alternative could be to create different sized messages (sensor_accel_fifo8, sensor_accel_fifo16, sensor_accel_fifo32, etc).
 - include helper script to unroll logged sensor_accel_fifo/sensor_gyro_fifo

dagar, 2020-04-08:
Yet another possible alternative, we could log per axis with a new message of perhaps 128 entires (uint16).

bkueng, 2020-04-09:
When I tested this, logger was easily able to keep up with 1khz, depending on overall CPU load you want to increase its priority though.
The current fifo messages are definitely too large, we can do it if we reduce them to 8 elements @ 1khz (or combine in some other manner).

BTW flight review already shows the messages.

dagar, 2020-07-10:
Done.

closed 2020-07-10 by dagar