PX4/PX4-Autopilot #14649: param_modify_on_import breaks calibration on reboot

navioglider, 2020-04-12:
**Describe the bug**
i have an couple sensors that i can calibrate, but the calibration disappears after a reboot. i traced the problem back to the function "param_modify_on_import" which modifies the perfectly good parameters CAL_XXX_ID and causes the problem. my guess is that it's related to #14572?

**To Reproduce**
1. calibrate e.g. mpu9250 gyroscope
2. restart px4, or reload parameters
3. watch how CAL_GYRO0_ID gets changed

**Expected behavior**
if "param_modify_on_import" does not change these parameters, then the calibration does not dissapear.

**Log Files and Screenshots**

> px4 starting.
> 
> INFO  [px4] Calling startup script: /bin/sh ../airframe.config 0
> INFO  [parameters] param modify: CAL_GYRO0_ID, value=0x240002 (old=0x240102)
> INFO  [parameters] param modify: CAL_GYRO1_ID, value=0x440002 (old=0x440302)
> INFO  [parameters] param modify: CAL_GYRO_PRIME, value=0x240002 (old=0x240102)
> INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
> mpu9250 #0 on SPI bus 0 (devid=0x1)
> lsm9ds1 #0 on SPI bus 0 (devid=0x3)
> lsm9ds1_mag #0 on SPI bus 0 (devid=0x2)
> lis3mdl #0 on I2C bus 1 (external)
> ms5611 #0 on I2C bus 1 (external)
> ms4525_airspeed #0 on I2C bus 1 (external)
> ERROR [sensors] fewer accels than calibrations, resetting all CAL_GYROx_ID
> ...

**Drone**
raspberry pi 4b + navio2

bkueng, 2020-04-14:
Thanks for reporting. Can you give https://github.com/PX4/Firmware/pull/14664 a test?

navioglider, 2020-04-14:
that fixed it, thanks!

closed 2020-04-14 by navioglider