PX4/PX4-Autopilot #14651: EKF2 replay on SITL logs is not reproducible

vanurag, 2020-04-13:
**Describe the bug**
I am using EKF2 replay to tune some of my parameters. As a starting point, I wanted to see if I can reproduce similar ekf2 output when I replay a log with unchanged parameters. I used jmavsim to perform a short flight session. I then performed ekf2 replay on this log. I am using Tools/ecl_ekf/process_logdata_ekf.py for generating analysis plots for the original log and the replayed log. I notice that they differ quite significantly.

**Log Files and Screenshots**
- [Original log file](https://review.px4.io/plot_app?log=407615c9-9d13-453b-b988-fbcd05d8f637)
- [Replayed log file](https://review.px4.io/plot_app?log=1ef81b28-cd83-4def-ada3-58995a109957)

**EKF analysis plots**
- [Original log analysis](https://github.com/PX4/Firmware/files/4469426/11_22_21.ulg.pdf)
- [Replayed log analysis] (https://github.com/PX4/Firmware/files/4469427/11_23_20_replayed.ulg.pdf)

dagar, 2020-04-17:
@bresch @jkflying @kamilritz FYI

vanurag, 2020-04-23:
Hey guys, any update on this issue? I saw that a new issue for [unit-testing EKF replay](https://github.com/PX4/Firmware/issues/14695) has been opened. Does that mean the EKF replay in it's current state is not functional?

priseborough, 2020-04-28:
I've run a check using a log generated using jmavsim built using master at commit https://github.com/PX4/Firmware/commit/326d8efc166331d5b9a0fc96836b68e82142a409 and it looks like there are two issues.

The first, which is easily fixed is that during replay, the logging of the uORB data does not start until after 20 seconds. This can be fixed by reducing the number of microseconds between subscription retries in logger.h, eg:

replace

static constexpr hrt_abstime TRY_SUBSCRIBE_INTERVAL{1000 * 1000};	// interval in microseconds at which we try to subscribe to a topic

with

// interval in microseconds at which we try to subscribe to a topic
#ifdef ORB_USE_PUBLISHER_RULES
static constexpr hrt_abstime TRY_SUBSCRIBE_INTERVAL {1000};
#else
static constexpr hrt_abstime TRY_SUBSCRIBE_INTERVAL {1000 * 1000};
#endif

The second issue appears to be an increase in innovations due to incorrect time stamping of input data resulting in GPS data being retrieved from the buffer and at the wrong time, eg:

These are the original logged GPS velocity innovations for the simulated flight which included an acceleration to the North followed by a a decel and stop, followed by an RTL on reciprocal course back to the takeoff location:

[image]

Here is the north acceleration value:
[image]

Small velocity innovations like this are common and normally caused by timing offsets and/or uncompensated offsets from IMU to GPS antenna combined with angular rates.  The magnitude of velocity innovations in the SITL log would be consistent with GPs being approx 70msec less delayed than the 110msec that the EKF2_GPS_DELAY parameter specifies. 

Replaying the log shows a much larger and opposite sign innovation levels during accel/decel:

[image]

The sign and amplitude of the North velocity innovation during the initial acceleration phase (before the filter 'corrects' the states) is consistent with an additional 200msec (1 GPS frame) delay, eg:

[image]

What this doesn't establish is whether the timing error is present in the original SITL log or introduced by the replay. Plotting the GPS and EKF north velocity from the original SITL log shows the GPS to be delayed relative to the EKF by over 350 msec according to its own timestamp, eg:

[image]

This suggests that the issue is caused by the timestamping of GPS data when it is logged in SITL

priseborough, 2020-04-28:
The next step is to determine if this is a jmavsim issue of if it also affects logs generated using gazebo

priseborough, 2020-04-30:
Gazebo has a GPS delay rel to the time stamp of around 110msec

[image]

Here is the same code and post  takeoff manoeuvre flown using jmavsim which shows a 200msec delay

/home/parallels/src/Firmware/build/px4_sitl_default/tmp/rootfs/log/2020-04-30/08_48_00.ulg

It is looking like this GPS time stamp problem is specific to jmavsim

priseborough, 2020-05-02:
Replaying these generated gazebo and jmavsim files and comparing GPS velocity innovations with the original SITL log using master at commit 0c8dcf94bc256801a45475cdd7bc6490e9fb8af7 gave the following:

jmavim results generated using following velocity profile:
[image]

GPS velocity innovation from original jmavsim log - these look high and are consistent with the observed excessive lag of the GPS relative to truth data:
[image]

GPS velocity innovations from replayed jmavsim log - these are higher than the original log and indicate additional GPS timing delay:
[image]

Gazebo results generated using following velocity profile:
[image]

GPS velocity innovations from original gazebo log - these look good and indicate the time stamping of the GPS data received by the EKF is correct:
[image]

GPS velocity innovations from replayed gazebo log - these are higher than the original log and indicate additional timing delay has been introduced:
[image]

In summary there are two error sources:

- jmavsim has excessive GPS delay
- The processing chain from logging of SITL data to replay of the same data has introduced what appears to be a one sample (approx 200msec) time delay in the GPS data relative to the IMU data.

The next step is to obtain flight test data using the same branch which will indicate if the additoinal GPS delay seen during replay is still present. If not then the focus of the investigation will be on the time stamping of the GPS data during logging of SITL data.

priseborough, 2020-05-02:
Replay of a flight log generated from an earlier commit with compatible uORB messages showed that the effect was still present.

priseborough, 2020-05-05:
The time offset during replay also affects Baro data and is consistent with the GPS offset, eg see correlation between vertical innovations and climb rate that is only present in replay:

Original Log:
[image]

Replayed Log:
[image]

Magnetometer data as evidenced by innovations during a yaw spin also shows evidence of an offset that is consistent with a time stamp offset between 100and 200 msec:

Original Log:
[image]

Replayed Log:
[image]

Given the number of sensor types affected, a replay bug affecting the time stamping of IMU data is possible.

priseborough, 2020-05-14:
@bkueng Are you aware of any changes to replay that could be causing the sensor timing offset?

bkueng, 2020-05-18:
No. Is that in simulation only?

jmavsim gps delay is explicitly set to 200: https://github.com/PX4/jMAVSim/blob/master/src/me/drton/jmavsim/Simulator.java#L452. We can reduce that.

priseborough, 2020-05-28:
It seems to be affecting replay of both SITL and flight logs. I will confirm the latter with some data gathered using latest master.

priseborough, 2020-06-09:
Replay is showing a large time slip (accumulated IMU delta time not matching lapsed time)

[image]

priseborough, 2020-06-09:
This positive gradient indicates that when replay is running, not all IMU data is being received or processed by the ekf2

priseborough, 2020-06-09:
Running replay with a printf in ekf2_main.cpp to monitor the time difference between subsequent IMU data, eg:

static int64_t last_imu_time = 0;
printf("imu_time_delta_us=%i\n",(int64_t)imu_sample_new.time_us - last_imu_time);
last_imu_time = (int64_t)imu_sample_new.time_us;

Shows that frequently data is skipped and sometimes duplicated, eg:
imu_time_delta_us=8000
imu_time_delta_us=4000
imu_time_delta_us=4000
imu_time_delta_us=4000
imu_time_delta_us=4000
imu_time_delta_us=8000
imu_time_delta_us=4000
imu_time_delta_us=4000
imu_time_delta_us=8000
imu_time_delta_us=4000
imu_time_delta_us=4000
imu_time_delta_us=4000
imu_time_delta_us=4000
imu_time_delta_us=4000
imu_time_delta_us=4000
imu_time_delta_us=4000
imu_time_delta_us=4000
imu_time_delta_us=8000
imu_time_delta_us=0
imu_time_delta_us=4000
imu_time_delta_us=4000
imu_time_delta_us=8000
imu_time_delta_us=4000

priseborough, 2020-06-11:
What I have been able to determine so far is that within 

ReplayEkf2::handleTopicUpdate 

the ekf2_timestamps data  retrieved from the file using 

memcpy(&ekf2_timestamps, data, sub.orb_meta->o_size);

is missing frames. I checked the ekf2_timestamps  data published to the uORB during SITL operation and there are no frames missing.

priseborough, 2020-06-11:
Converting the SITL generated log file to csv and inspecting shows that the one frame gaps in the ekf2_timestamp data are present in the file, eg:

[image]

Investigation will now ficus on the logger

priseborough, 2020-06-17:
A bench test on a F7 board (Pixhawk4) shows that there are no missing samples, however the baseline EKF update rate for this hardware is 200Hz, eg:

[image]

priseborough, 2020-06-18:
Closing this as it has been picked up here: https://github.com/PX4/Firmware/issues/15088

I have confirmed with flying and replay today that with current upstream master, the logging defect does not affect the F7 hardware i fly with.

closed 2020-06-18 by priseborough

bkueng, 2020-06-18:
Thanks @priseborough for clarifying. I'll fix the underlying issue in SITL.

vanurag, 2020-06-18:
Thanks @priseborough for looking into the issue extensively. I can confirm that as per #15088, just by polling on ekf2_timestamps for now, I could get much better results in SITL.