PX4/PX4-Autopilot #14654: one question from attitude controller

Biffilion, 2020-04-13:
I use the code with v1.9.2.
In the file mc_att_control_main.cpp.I know to use the cascade controller here.But why control_attitude_rates before control_attitude?
I think the angular acceleration controller uses the result of the last angular velocity controller.
Why not run the angular velocity controller first?

julianoes, 2020-04-14:
High level, for a good control performance the delay from the moment a sensor value is measured until the moment the motor control is sent out is crucial. The smaller this delay, the better (and easier) is the control problem. This is especially true and essential for the attitude rate loop while the other cascaded controllers are less critical.

I believe that's the reason why on a new sensor value, the rate loop runs immediately, so as early as possible and the other controllers later. The other controllers will still produce a rate setpoint but it won't get applied until the next iteration.

dagar, 2020-04-17:
> Why not run the angular velocity controller first?

Please update to newer PX4 (eg v1.11.0-beta0) and take a look at the controllers and message architecture. The multicopter rate controller is now a standalone module and the primary gyro is filtered and published as `vehicle_angular_velocity`. This runs as the highest priority thread in the system to minimize end-to-end latency and jitter.

closed 2020-04-17 by dagar