PX4/PX4-Autopilot #14655: NuttX mkfatfs doesn't seem to work

dagar, 2020-04-13:
I'm seeing this on all boards.

[image]

In many cases `umount /fs/microsd` fails, but regardless of a successful umount mkfatfs still fails with the same error.

mirkix, 2020-11-30:
@dagar @davids5  When using `mkfatfs -F 32 /dev/mmcsd0` it is working on a CubeBlack (STM32F4), only use the `-F 32`parameter. On o CubeOrange (STM32H7) it fails even with `mkfatfs -F 32 /dev/mmcsd0` but when I add `# CONFIG_STM32H7_SDMMC_IDMA is not set` to the NuttX defconfig, I can use it successful with a CubeOrange as well.

davids5, 2021-12-28:
fixed with https://github.com/PX4/NuttX-apps/pull/14

closed 2021-12-28 by davids5