PX4/PX4-Autopilot #14659: Unstable GPS data while using the 2nd telemetry on TELEM2 port - v1.11.0-beta1

saengphet, 2020-04-14:
**Describe the bug**
Hi, I found very strange GPS behavior after uploaded "v1.11.0-beta1" to pixhawk 4. Firstly, I configured MAV_1_CONFIG = TELEM2 (My UAV has been flown using this config. before since firmware v1.8.2, v1.9.2 and v1.10.2 and I got no error). Then, I investigated using"mavlink inspector" in QGC and take a look at GPS_RAW_INT. 

GPS signal seems unstable, the fluctuation ranges from 0 to 5 Hz. It should be a constant value or nearly constant (compared with an older firmware, I have already tested before).

Finally, the result is the flight mode could not be changed from ALTITUDE to POSITION, No Green GPS LED light visible. I managed the test as follows 

**To Reproduce / Test Case**
Case 1: v1.11.0-beta1 --> OK
MAV_1_CONFIG = Disabled
https://review.px4.io/plot_app?log=9764baca-ec06-40aa-b1ba-79719d0b75fc

Case 2: v1.11.0-beta1 --> Error / ubx msg warning
MAV_1_CONFIG = TELEM2
https://review.px4.io/plot_app?log=a670c429-592e-4e52-a200-b811bd45f14d

Case 3: v1.11.0-beta1 --> OK
MAV_1_CONFIG = TELEM/SERIAL4 
https://review.px4.io/plot_app?log=98b8c1ec-d492-48b5-823a-870af29e5686

Case 4:  v1.11.0-beta1 --> Error / ubx msg warning
MAV_2_CONFIG =  TELEM2  

Case 5: v1.11.0-beta1 --> OK
MAV_2_CONFIG =  TELEM/SERIAL4 

Case 6: v1.11.0-beta1 --> Error / ubx msg warning
MAV_1_CONFIG = TELEM/SERIAL4 
MAV_2_CONFIG =  TELEM2  

Case 7: v1.10.1 --> OK (compare to Case 2 v1.11.0-beta1. It is not the same result even config. is the same)
MAV_1_CONFIG = TELEM2
https://review.px4.io/plot_app?log=29bea816-da9d-4cd4-b042-c0fdb14e489d

The error is the same whether telemetry is available on that port or not, indoor or outdoor if I configure MAV_X_CONFIG to TELEM2. According to the flight log of Case 2, you will see the strange GPS behavior at "GPS Uncertainty and GPS Noise & Jamming". In every test case, Telemetry works fine.

However, GPS works fine if I upload the older firmware "v1.10.1" (v1.8.2 / v1.9.2 is the same) Is it firmware bug ? or Did I miss anything?

**Drone (please complete the following information):**
- Describe the type of drone: Tiltrotor using E-flite Convergence airframe
- Hardware: Pixhawk4 + GPS pixhawk 4 + Holybro telemetry

bkueng, 2020-04-21:
Thanks for the detailed report. I can reproduce this, also on master.

- Pixhawk 4 with GPS attached
- `param set MAV_1_CONFIG 102`
- `param set SER_TEL2_BAUD 57600`

The problem goes away when disabling TX DMA (`CONFIG_USART3_TXDMA`) on Telem2. @davids5 can you have a look?

mrpollo, 2020-04-22:
Thanks for reporting @saengphet

bkueng, 2020-04-27:
Another report: https://logs.px4.io/plot_app?log=1c41e67f-696b-4b76-a669-cf4cd42330b0
```
MAV_1_CONFIG, 102
SER_TEL2_BAUD, 38400
```

davids5, 2020-08-08:
I have Isolated the problem and will have a fix in next week.

davids5, 2020-08-10:
Please re-test with https://github.com/PX4/Firmware/pull/15528

closed 2020-08-10 by dagar