PX4/PX4-Autopilot #14661: Pixracer mag not working

julianoes, 2020-04-14:
My (as far as I know original) Pixracer doesn't work anymore. There does not seem to be a mag driver running.

```
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V4
FW git-hash: 588195f46c2142a674072148bf21557c4cb13b5c
FW version: 1.11.0 80 (17498240)
FW git-branch: master
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: ec417d7466666801e911e50d72766225ca7790a1
Build datetime: Apr 14 2020 09:17:34
Build uri: localhost
Toolchain: GNU GCC, 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
PX4GUID: 000100000000303833363335510d00300041
MCU: STM32F42x, rev. 3
[hardfault_log] Fault Log is Armed
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
nsh: rgbled_pwm: command not found
Board sensors: /etc/init.d/rc.board_sensors
reset done, 50 ms
ms5611 #0 on SPI bus 2 (devid=0x3d)
mpu9250 #0 on SPI bus 1 (devid=0x24)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [mavlink] mode: Normal, data rate: 20000 B/s on /dev/ttyS0 @ 921600B
WARN  [mavlink] hardware flow control not supported
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
ERROR [mavlink] offboard mission init failed (-1)
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
Board extras: /etc/init.d/rc.board_extras
INFO  [logger] logger started (mode=all)
```

JacobCrabill, 2020-04-14:
Maybe this is due to the new IMU drivers?  Has the compass now been split out from the accel/gyro in the MPU drivers?

dagar, 2020-04-14:
> Maybe this is due to the new IMU drivers? Has the compass now been split out from the accel/gyro in the MPU drivers?

We're not running the mpu9250 mag by default, it messes up the IMU sampling and becomes noticeable with all the raw data now available PX4 side.

This should be okay because the pixracer has another onboard mag (hmc5883 or lis3mdl). I recently fixed the internal hmc5883 start (https://github.com/PX4/Firmware/pull/14634). 

@julianoes any idea what's physically on your particular board? You could try starting the hmc5883 and lis3mdl with debug enabled. 

https://github.com/PX4/Firmware/blob/a1ad7aa626b27f1cab573b134102b2a58085615e/boards/px4/fmu-v4/init/rc.board_sensors#L27-L32

julianoes, 2020-04-15:
```
nsh> hmc5883 -T -s -R 2 start 
nsh> listener sensor_mag
never published
nsh> lis3mdl -s start 
nsh> 
nsh> 
nsh> listener sensor_mag
never published
nsh> 
```
[image]

dagar, 2020-05-06:
@julianoes can you tell what's physically on that board? Could you check with current master? 
I'm wondering if this is a factor? 

https://github.com/PX4/Firmware/blob/9963bb6c40925e5facf183045e1fc359b9e7d959/boards/px4/fmu-v4/init/rc.board_sensors#L8-L14

julianoes, 2020-05-07:
@dagar as far as I remember, this is an original Pixracer but maybe an early one.

The problem is still the same:

```
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V4
FW git-hash: 41660c355035192f8c7e562db97a153d6e50482b
FW version: 1.11.0 80 (17498240)
FW git-branch: master
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: 66b4f2c4f2128994f8e8a908d4888f6d37565cfd
Build datetime: May  7 2020 09:07:43
Build uri: localhost
Toolchain: GNU GCC, 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
PX4GUID: 000100000000303833363335510d00300041
MCU: STM32F42x, rev. 3
[hardfault_log] Fault Log is Armed
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
nsh: rgbled_pwm: command not found
Board sensors: /etc/init.d/rc.board_sensors
reset done, 50 ms
ms5611 #0 on SPI bus 2 (devid=0x3d)
mpu9250 #0 on SPI bus 1 (devid=0x24)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [mavlink] mode: Normal, data rate: 20000 B/s on /dev/ttyS0 @ 921600B
WARN  [mavlink] hardware flow control not supported
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
Board extras: /etc/init.d/rc.board_extras
INFO  [logger] logger started (mode=all)
```

I assume this Pixracer only has the MPU9250 and no HMC5983. I'm not sure how many of these are out there.

julianoes, 2020-06-14:
I've tested two Pixracers (one unknown, and one labelled HobbyKing rev. 2) both of which worked.

@bkueng also found one Pixracer which did not work, presumably both are very early samples.

dagar, 2020-06-14:
> @bkueng also found one Pixracer which did not work, presumably both are very early samples.

Which mag is physically on that board?

julianoes, 2020-06-14:
Mpu9250

dagar, 2020-06-15:
All pixracers I've seen in person have a dedicated mag (hmc5883 or lis3mdl).

If there's a valid case with no dedicated on board mag what we could do is optionally start the mpu9250 with the mag enabled (-M) in the startup script if the others fail. https://github.com/PX4/Firmware/blob/6c78c62d9d3a016634a6f32d5190c2a679272adf/boards/px4/fmu-v4/init/rc.board_sensors#L19-L24

Having the mpu9250's i2c slave ak8963 enabled does make a mess of the scheduling, but we're more equipped to deal with it now that nearly all of the IMU processing is downstream and actually computes the real output data rate from published data rather than trusting the sensor.

mrpollo, 2020-08-31:
is this still blocking the PX4 v1.11 release @dagar @julianoes ?

julianoes, 2020-09-01:
So it's still not working for me. However, this is probably because I have an older/prototype Pixracer.

We can ignore this but it means @bkueng and I have a pixracer each for the graveyard.

closed 2020-09-01 by julianoes

dagar, 2020-09-01:
> So it's still not working for me. However, this is probably because I have an older/prototype Pixracer.
> 
> We can ignore this but it means @bkueng and I have a pixracer each for the graveyard.

If you think it's important then I'd add a bit of logic in fmu-v4's rc.board_sensors to start the mpu9250 with mag (-M) if both the lis3mdl and hmc5883 fail to start.

bkueng, 2020-09-02:
Only if it's little effort, it's not that important to me.