PX4/PX4-Autopilot #14670: BAT_V_DIV and BAT_A_PER_V params missing unless battery_status module is included in build

modaltb, 2020-04-14:
**Describe the bug**
Our hardware doesn't use the `battery_status` module, rather a power module connected over I2C, so we aren't including `battery_status` in our cmake build file.  I believe that without including `battery_status` in the build, you fail to pick up the parameters in `analog_battery_params_deprecated.c`.  

I see this PR from sometime ago, which is in the same area: https://github.com/PX4/Firmware/pull/13299/commits/c20190640039be4379b9fc2e5a0ec160d5b68699 .  It's using param_find to 'force them into the list' but I don't think this is working.

**To Reproduce**
Perhaps somewhat unique, but:
1. build a target without the `battery_status` module
2. Connect to QGC and open the Power settings
3. Observe the "Missing Params:.. BAT_V_DIV .... BAT_A_PER_V". warning

If you have v5x hardware (or ModalAI FCv1), you might be able to replicate pretty easy loading a master branch build and connected to QGC...  I don't think the v5x has `battery_status` either

**Expected behavior**
No warning in QQC from these legacy params being missing.  I can add them to our target or in the voxlpm driver but seems hacky.

**Log Files and Screenshots**

[image]

**Drone (please complete the following information):**
- modalai_fc-v1

**Additional context**
Simple work around is for me to add these params to the voxlpm driver, but seems too hacky...

julianoes, 2020-04-20:
>  Simple work around is for me to add these params to the voxlpm driver, but seems too hacky...

I would argue that's what is needed for backwards compatibility.

Also, see how the deprecated params are synced with the new ones here:
https://github.com/PX4/Firmware/blob/6e18cb85b1911ac052838c73a6fbb9b8d6f72d2c/src/modules/battery_status/analog_battery.cpp#L92-L97

So if you can make use of the [battery lib](https://github.com/PX4/Firmware/blob/master/src/lib/battery/battery.h) and copy the `migrateParam` part, and include both param definition files, you should be good.

modaltb, 2020-04-20:
OK great, thanks for the tips, I can move forward with this easy enough!

closed 2020-05-04 by modaltb