PX4/PX4-Autopilot #14671: Arm tone alarms after pressed pre-arm button to disarm vehicle in fixed-wing mode-tiltrotor-v1.11.0-beta1

saengphet, 2020-04-15:
**Describe the bug**
I found this issue during the tiltrotor flight test. After pre-arm button at GPS pixhawk 4 was pressed to disarm vehicle in fixed-wing mode, buzzer alarmed "long beep (arm tone)" that is the sound of "arm state" instead of "double short beep (disarm tone)". This is so weird and makes me scared of motor spinup unintentionally (but motors did not spinup).

**To Reproduce**
1. Press pre-arm button at GPS to make vehicle ready for arming, then arm vehicle in multirotor state via RC transmitter. Motors spinup at idle PWM.
2. zero RC throttle, then activate transition switch to fixed-wing. Motors stop spinning, tilt-servos move down to horizontal position.
3. At fixed-wing state, press "pre-arm button again" to disarm vehicle. (because I don't want to go back to multirotor state again before disarming vehicle)
4. You will hear the "disarming tone as double short beep" follow by "arming warning tone as long beep" suddenly. No motor spinning, tilt-servos t return to disarm position (vertical).
 
ref. of tune meanings
https://docs.px4.io/v1.9.0/en/getting_started/tunes.html

video link:
https://drive.google.com/file/d/13HWmceGjHeR9kE2br4Sv5yqyTqhRS3Sb/view?usp=sharing

**Expected behavior**
After press pre-arm button to disarm vehicle in fixed-wing mode, "double short beep" alarms to notice user that vehicle is going to be in disarm state. No "arm tone" until vehicle is armed by RC transmitter or QGC.

**Drone (please complete the following information):**
Describe the type of drone: Tiltrotor using E-flite Convergence airframe
Hardware: Pixhawk4 + GPS pixhawk 4 + Holybro telemetry

dagar, 2020-04-17:
Thanks for reporting.

dagar, 2020-05-06:
Would you be able to share a log file (.ulg) from boot (SDLOG_PROFILE) showing this problem?

saengphet, 2020-05-07:
> Would you be able to share a log file (.ulg) from boot (SDLOG_PROFILE) showing this problem?

OK, I will test it again to get the log from boot.

LorenzMeier, 2021-06-02:
We have reworked this since, closing.

closed 2021-06-02 by LorenzMeier