PX4/PX4-Autopilot #14675: Some AUX pwm outputs go lower on arm

Thijs----, 2020-04-15:
**Describe the bug**
When the UAV is armed (via RC), some AUX pwm outputs (actuator_outputs_1) shortly go to a lower value, although this is not commanded via an actuator_control message. It seems that this only happens with AUX outputs who's PWM_AUX_DIS is set.

**To Reproduce**
Steps to reproduce the behavior:
1. change PWM_AUX_DIS2 (or another) from -1 to 100
2. (set SDLOG_MODE 0 -> 1)
3. arm UAV via onboard arm-switch
4. arm via RC
5. see in log: during 2nd arming the actuator_outputs_1[1] goes down for a short while

**Expected behavior**
The output should be following the commanded value via the mixer file, thus stay constant.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- 1st log: modified firmware [PX4 Flight Review](https://logs.px4.io/plot_app?log=45d2537b-d485-4103-9a95-22cfb1819fef), [AUX output chart](https://logs.px4.io/plot_app?log=45d2537b-d485-4103-9a95-22cfb1819fef#Nav-Actuator-Outputs-AUX).
PWM_AUX_DIS2 and PWM_AUX_DIS3 set to 100 (AUX output 1 & 2).
Note that the lower value is 1050 and ramps up to commanded value.
Used mixer: 
actuator_outputs_1[0] = actuator_control_0[7] (landing gear)
actuator_outputs_1[1] = actuator_control_2[1] (mount pitch)
actuator_outputs_1[2] = actuator_control_2[4] (mount custom)

- 2nd log: PX4 v1.10.0, vmount enabled [PX4 Flight Review](https://logs.px4.io/plot_app?log=d626193a-b86e-4510-a230-463f0c6caed6), [AUX output chart](https://logs.px4.io/plot_app?log=d626193a-b86e-4510-a230-463f0c6caed6#Nav-Actuator-Outputs-AUX).
PWM_AUX_DIS2 set to 100 (AUX output 1).
Note that the lower value goes to 100, then 1050 and ramps up to commanded value.
Used mixer: mount activated

- 3rd log: PX4 v1.10.0, default mixer [PX4 Flight Review](https://logs.px4.io/plot_app?log=20112ab9-12fe-4f15-96a0-fb3691030048), [AUX output chart](https://logs.px4.io/plot_app?log=20112ab9-12fe-4f15-96a0-fb3691030048#Nav-Actuator-Outputs-AUX).
PWM_AUX_DIS2 set to 100 (AUX output 1).
Note that the lower value goes to 100, then 1050 and ramps up to commanded value.
Used mixer: default (pass trough?)

Add screenshots to help explain your problem.
1st log:
[image]
*Note that the arming_state is scaled
2nd and 3rd log:
[image]
*Note that the arming_state is scaled

**Drone (please complete the following information):**
- octo/quadcopter
- PX4 1.10.0

**Additional context**
Add any other context about the problem here.

dagar, 2020-04-17:
Could you please re-test this on the current v1.11.0-beta1 (or master)? If it's still an issue I'd like to make sure it's fixed for the upcoming v1.11.0 stable release.

Thijs----, 2020-04-20:
I Tested with v1.11.0-beta1: [PX4 Flight Review](https://logs.px4.io/plot_app?log=6a80d3e8-fce1-4aba-beaf-6712631d0fb2) [AUX output chart](https://logs.px4.io/plot_app?log=6a80d3e8-fce1-4aba-beaf-6712631d0fb2#Nav-Actuator-Outputs-AUX)
Same effect: with PWM_AUX_DIS2 set too 100, on arm output 1 goes down to 100, then 1050 and ramp up to 1500 again.

Thijs----, 2020-04-21:
Two updates:
1. I also tested with PWM_AUX_DIS2 set to 1900: on arm AUX output 1 goes shortly to 1900 and direct back to commanded value: [PX4 Flight review](https://logs.px4.io/plot_app?log=9a1fbc52-6b42-45c6-840e-d09c2b82f871). So same behavior, except there is no ramp up from 1050.

2. I also noticed during another test that if even no PWM_AUX_DIS is set, this behavior occurs. In this [PX4 Flight review](https://logs.px4.io/plot_app?log=300dd318-66e0-423a-8393-60f48db6dd91) the AUX output 0 is commanded to 1000, on arm it goes shortly to 1500 (default disarm pwm).

JaCoHSU, 2024-07-24:
@Thijs---- I am facing the same problem. Is there any solution?

Thijs----, 2024-07-26:
Hi @JaCoHSU, I don't think there was a solution or workaround.
I can't verify it, because I no longer have access to the project.