PX4/PX4-Autopilot #14677: Multicopter v1.11 beta testing - max throttle is suddenly asserted

dagar, 2020-04-15:
https://logs.px4.io/plot_app?log=23c4f03d-819d-4f6e-a9fa-f6a8fdc57e01

dagar, 2020-04-15:
The cause of the "throttle jump" appears to have been caused by the altitude estimate jumping when switching back to baro aiding. 

[image]

[image]

What's the motivation for using `EKF2_HGT_MODE = GPS`?

dagar, 2020-04-15:
I think there must be a bug within ecl/EKF where the height offset is being handled correctly as you switched between range aid, gps height, and baro height.

https://github.com/PX4/ecl/blob/6f383248cafa4542d1be6cd884e37783af270b54/EKF/control.cpp#L1027-L1059

dagar, 2020-04-15:
I believe the problem is the ordering of `Ekf::resetHeight()` triggered by `Ekf::controlHeightSensorTimeout()` where the `_hgt_sensor_offset` is only updated later by `Ekf::controlHeightFusion()`.

limhyon, 2020-04-16:
Right. We have GPS/range(aid)/baro. I think there are bugs in ECL/EKF regarding switching between main source of height.

dagar, 2020-04-16:
Fixed in https://github.com/PX4/ecl/pull/799 thanks to @bresch 

PX4/Firmware PR for testing https://github.com/PX4/Firmware/pull/14685

bresch, 2020-04-16:
@dagar I'm able to reproduce the issue in SITL. The ECL fix solves most of the issue (the reset is now much smaller), but the real cause of the thrust step is because the reset isn't handled properly in the flight task: the reset handler is called, but the internal states aren't reset correctly.
I think I'll have to add a functional test to fix it and to make sure that it doesn't break again.

closed 2020-04-22 by bresch