PX4/PX4-Autopilot #14688: TiltRotor VTOL goes to 0 thrust during transition under v1.10.1

bys1123, 2020-04-16:
**Describe the bug**
We are developing a VTOL from 2 years ago, it runs will under 1.7.3.
Video is here: https://www.youtube.com/watch?v=am2847IQGLU
Recently we are updating the firmware follow upstream PX4.
But it will has 0 thrust during translation under 1.10.1.

**Expected behavior**
Smooth transition.

**Log Files and Screenshots**

[image]
https://logs.px4.io/plot_app?log=79824200-9462-4298-80c4-d8fde13c9e01

**Drone (please complete the following information):**
TiltRotor VTOL 1145mm wingspan

I'm not sure this is a bug or we missing some parameter, we already tried several days, but didn't fix it. So we need some help on this, thanks!

dagar, 2020-04-17:
@bys1123 if possible please test the current v1.11.0-beta1. It's still possible to get this fixed in time for the v1.11.0 stable release (if it's still broken in current master).

bys1123, 2020-04-17:
@dagar Thanks! I'll test the current master in three days.

sfuhrer, 2020-04-22:
@bys1123 The altitude drop is a known issue. It only happens when you are flying in stabilized mode and are ascending fast. When you then switch into transition, the altitude controller that is running in then tries to immediately and aggressively hold the altitude and cuts throttle strongly. It's much smoother when you already before the transition were flying in altitude or position control mode.
In your case the drop in altitude was pretty big because your vehicle has a very high hover throttle that it needs to even fly level (around 68% looking at the log). That means that the vehicle has a small control thrust margin, and it only very slowly can accelerate upwards again after it has begun to sink (it keeps descending for a couple of seconds even with max hover throttle):
[image]
Given that you can't change the hardware (what would be the best thing!) and improve the hove thrust margin (a healthy system has a hover throttle in the range of 40-60%), I would advise you to set:
MPC_THR_HOVER to 65
MPC_THR_MIN to 40

bys1123, 2020-04-23:
Thanks! We tested with this parameter, works better now! @sfuhrer 
https://logs.px4.io/plot_app?log=a3e8367a-cd6b-4f7f-b5a1-b5c4cec74bb9

lydoraemon, 2020-06-16:
wow

lydoraemon, 2020-06-16:
New test in Mission mode: the throttle seems correct.

lydoraemon, 2020-06-16:
file:///C:/Users/Administrator/Desktop/%E5%8F%82%E6%95%B0%E5%9B%BE/bokeh_plot.png
New test in Mission mode: the throttle seems correct.

lydoraemon, 2020-06-16:
[image]
New test in mission mode: the throttle goes well.

closed 2020-07-03 by bys1123