PX4/PX4-Autopilot #14696: Unable to connect gazebo simulator with px4_sitl when launched with ROS

SarthakGupta22, 2020-04-18:
Ran the following commands in the terminal

no_sim=1 make px4_sitl gazebo
roslaunch gazebo_ros empty_world.launch world_name:=$(pwd)/Tools/sitl_gazebo/worlds/iris.world

The sitl could not connect to the simulator and the following is the description of the error

... logging to /home/robosteps/.ros/log/2c327300-8152-11ea-a0e1-dcf5056f1ce3/roslaunch-robosteps-29306.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://robosteps:36875/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [29316]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2c327300-8152-11ea-a0e1-dcf5056f1ce3
process[rosout-1]: started with pid [29327]
started core service [/rosout]
process[gazebo-2]: started with pid [29330]
process[gazebo_gui-3]: started with pid [29336]
[ INFO] [1587200052.789992503]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1587200052.794227938]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1587200052.795072001]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1587200052.797933138]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1587200053.419142731]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1587200053.467195565]: Physics dynamic reconfigure ready.
[ INFO] [1587200053.755808193]: Camera Plugin: The 'robotNamespace' param was empty
[ INFO] [1587200053.761385214]: Camera Plugin (ns = iris)  <tf_prefix_>, set to "iris"
[ INFO] [1587200053.766170480]: Camera Plugin: The 'robotNamespace' param was empty
[ INFO] [1587200053.770646776]: Camera Plugin: The 'robotNamespace' param was empty
[ INFO] [1587200053.773262954]: Camera Plugin (ns = iris)  <tf_prefix_>, set to "iris"
[ INFO] [1587200053.777746604, 0.004000000]: Camera Plugin (ns = iris)  <tf_prefix_>, set to "iris"
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'

Jaeyoung-Lim, 2020-04-18:
@SarthakGupta22 Have you checked out the instructions in http://dev.px4.io/master/en/simulation/ros_interface.html#launching-gazebo-with-ros-wrappers ?

SarthakGupta22, 2020-04-24:
Thanks @Jaeyoung-Lim and @pravardhanreddy, got this issue resolved. Missed sourcing the bash file, hence the error was coming.

closed 2020-04-24 by SarthakGupta22