PX4/PX4-Autopilot #14697: Linux Boards

bilginmu, 2020-04-18:
Is there any documantation about that how to run PX4 on linux boards ?

bilginmu, 2020-04-18:
I have Arm microprocessors Cortex-A9 on which is running real time linux.

hamishwillee, 2020-04-20:
Yes, see 
- https://dev.px4.io/master/en/setup/building_px4.html#raspberry-pi-23-boards
- https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html#raspberry-pi-hardware

I am closing this because this area is for issues/bugs, not support requests. If you wish to continue the discussion, please raise on https://discuss.px4.io/ (cross linking to here)

closed 2020-04-20 by hamishwillee

bilginmu, 2020-05-01:
Sorry for wrong area and thanks for reply but I have Petalinux and how to build for petalinux? Building px4 for petalinux and rasperry pi are the the same ?