PX4/PX4-Autopilot #14701: Automatic failsafe land in fixed wing RTL not possible

MaEtUgR, 2020-04-19:
**Expected behavior**
Goal: When losing RC the fixed-wing plane should fly back to home, loiter over it for a certain time and then slowly descend while loitering until it reaches the ground. This allows configuring a least possible damage failsafe for the worst case that the pilot loses control but didn't plan any landings.

By default a plane comes back in RTL and loiters indefinitely over the home point because `RTL_LAND_DELAY` is set to -1 in the fixed-wing defaults. This also works fine in SITL and the real world and saved my ass before.

**Describe the bug/To Reproduce**
I did some **fixed-wing RTL SITL** tests before risking my real plane:
1. setting `RTL_LAND_DELAY` to 0 presuming I already lost control completely when I don't regain connection until the plane reaches home and it should start descending immediately. It immediately upon reaching home starts descending going straight ahead without turning. After half the way to the ground, it suddenly starts ascending indefinitely continuing straight with the same heading. First thing in the log.
1. setting `RTL_LAND_DELAY` to 20 presuming I anyways want to loiter at least one full turn above my head before descending for good practice. It starts loitering and continues indefinitely. The many turns above home in the log.
1. The only way to get the plane down in auto but triggered manually which is not usable as a failsafe is to do an RTL first having it loiter above home. Then issue `commander land` on the console since QGC doesn't show the button or the mode in the mode switcher. The plane starts descending, doesn't loiter nicely around home anymore but rather comes down in a random spiral further away from home. At the end of the log.

**Log Files and Screenshots**
Here's the SITL log of my tests: https://logs.px4.io/plot_app?log=3cbedcc2-3823-409f-84da-65ab0d88df6d

Showing 1. visually.
[image]

**Drone (please complete the following information):**
- Master state ffa9fdbd5677eaedf0bec9bea99967cfaee9e562
- SITL `make px4_sitl gazebo_plane`

**Additional context**
Sounds like it's related to https://github.com/PX4/Firmware/issues/12035

hamishwillee, 2020-04-19:
So just to be clear, what we're seeing here sounds like the documented behaviour: http://docs.px4.io/master/en/flight_modes/return.html#fixed-wing-fw

I think you're saying this is not optimal. Currently RTL_LAND_DELAY=-1 means  "do not land/loiter" forever" and you're saying it should mean "loiter while descending slowly"? Sounds good to me. 

If you change this, can we keep the docs in sync - e.g. note what has changed, especially any configurable params.

Antiheavy, 2020-04-19:
> I think you're saying this is not optimal. Currently RTL_LAND_DELAY=-1 means "do not land/loiter" forever" and you're saying it should mean "loiter while descending slowly"? Sounds good to me.

@hamishwillee I don't think you are correct here.

@MaEtUgR It sounds like you are describing two things: 1) bugs with how RTL_LAND_DELAY works, and 2) new landing logic for the RTL around home scenario where the vehicle loiters slowly to the ground if RTL_LAND_DELAY is 0 or a positive value?

I'm curious: What happens if throttle is forced to 0 when in a fixed wing loiter?  Will the position controller maintain the loiter ground-track nicely even though the vehicle won't maintain altitude?  I know in Mission mode the vehicle does things like spiral up/down to reach an altitude setpoint ignoring the mission's ground track.

hamishwillee, 2020-04-20:
@Antiheavy Thanks. I'm not at all sure, hence the question :-)

MaEtUgR, 2020-04-20:
> I think you're saying this is not optimal. Currently RTL_LAND_DELAY=-1 means "do not land/loiter" forever" and you're saying it should mean "loiter while descending slowly"?

@hamishwillee I think the default is ok or could be discussed separately based on user feedback. I just expect to have the option to spiral down slowly because RTL is the default failsafe for RC loss and if I'm unable to regain control e.g. remote broken or ran out of battery I'd expect it to do what I consider the least damage reaction given no further planning.

> 1) bugs with how RTL_LAND_DELAY works

@Antiheavy Yes, big bug that it starts ascending again while landing with `RTL_LAND_DELAY` 0 this leads not just to a crash but a flyaway. And `RTL_LAND_DELAY` > 0 doesn't start landing.

> 2) new landing logic for the RTL around home scenario where the vehicle loiters slowly to the ground

I wasn't aware it's something new. I thought that's the failsafe when auto-land is enabled in RTL because it's the easiest way to land close to home. Is the originally expected use case landing on a runway planned through a mission? https://docs.px4.io/master/en/flight_modes/return.html#fixed-wing-fw What do you think of the feature then? Do you plan the landing using a mission for every manual flight?