PX4/PX4-Autopilot #14705: prearm with linux

navioglider, 2020-04-19:
**bug description**
setting COM_PREARM_MODE="Always" does not seem to do anything when running linux (same behaviour as if "Disabled"). From this comment in the code it seems like you already know about the issue:
https://github.com/PX4/Firmware/blob/7c533e5a535e48b460dda1a4f148b830155f3e06/src/drivers/linux_pwm_out/linux_pwm_out.cpp#L360-L362
(although it has been like that for a looong time)
reactivating line 362 seems to me to make prearm work again and not break anything. did the bug get fixed at some point?

**drone**
raspberry pi 4b + navio2

dagar, 2020-04-19:
@navioglider this isn't a direct answer to your question, but `linux_pwm_out` was never on par with the "normal" output modules (eg `pwm_out`) we use on common boards like the pixhawk. My plan for the navio2 was to implement the same low level PWM functions we use on NuttX, then use the regular `pwm_out` module and purge linux_pwm_out.

Here's an old branch I had started (https://github.com/dagar/PX4-Firmware/commits/pr-linux_pwm_out_cleanup). Look for `boards/emlid/navio2/pwm/pwm_servo.cpp`.

In terms of overall PX4 platform unification (Linux & NuttX) the output module and init scripts are the largest remaining items (for Navio2 at least). If we can migrate linux_pwm_out -> pwm_out, and the init system to what's in `ROMFS/` we'll be in a much better position to maintain Linux board support going forward.

FYI @SalimTerryLi

dagar, 2020-04-19:
Direct link to what needs to be implemented for Navio2.

https://github.com/dagar/PX4-Firmware/blob/4cd633fcb19e809835fca99e3768d6aa701aeac1/boards/emlid/navio2/pwm/pwm_servo.cpp#L10-L88

[image]

SalimTerryLi, 2020-04-19:
That is. Thanks for mentioning that. The first thing that come to my mind is whether can I deploy PCA9685 in this way or not. I will try to figure out that. But I cannot provide any useful info that would benefit this issue, as I have no Navio2 HAT to do any test.

Onboard PWM channels should be managed in this way, for sure.

dagar, 2020-04-19:
Related: https://github.com/PX4/Firmware/pull/14707