PX4/PX4-Autopilot #14717: Thermal Calibration corrupts accelometers data to show wrong horizon roll

MartinStokelj, 2020-04-21:
**Describe the bug**
After enabling thermal calibration and calibrating the accelometers we can see horizon off for about 15 degrees roll angle.
Autopilot has 2 accelometers, LSM303D,MPU6000 and it looks like thermal calibration for both of them is somehow calculated wrong (horizon showing roll for about 15 degrees and after level horizon calibration we cant calibrate anything more, always getting error wrong direction detected).
After removing LSM303D from the board the thermal calibration gave good result since horizon was almost in level after calibrating accelometers.
Tryed on FW version 1.10.0 and 1.10.1

**To Reproduce**
Steps to reproduce the behavior:
1. Set VTOL DeltaQuad airframe and enable thermal calibration
2. Do the thermal calibration
3. Calibrate accelometers and you will see wrong roll on horizon
4. Do the level horitzon calibration, the horizon is now ok
5. Do the accelometers calibration again, here we get error wrong direction detected

**Expected behavior**
Thermal calibration should not corrupt the accelometer data.

**Log Files and Screenshots**
[Params with thermal calibration done] https://we.tl/t-5RIMulHMc5

**Drone (please complete the following information):**
Stack: PX4 Pro
Vehicle: VTOL
Version: 1.10.0 
Git Revision: 4f6faac2c8000000

**Additional context**

dagar, 2020-04-21:
Do you have the thermal calibration log and the generated pdf?

MartinStokelj, 2020-04-21:
> Do you have the thermal calibration log and the generated pdf?

If you give me info how to create them I will.

dagar, 2020-04-21:
I'm referring to the offboard calibration method described here. https://docs.px4.io/master/en/advanced_config/sensor_thermal_calibration.html#offboard_calibration

MartinStokelj, 2020-04-21:
> I'm referring to the offboard calibration method described here. https://docs.px4.io/master/en/advanced_config/sensor_thermal_calibration.html#offboard_calibration

I will make offboard calibration tomorow and give the new feedback.

MartinStokelj, 2020-04-22:
> I'm referring to the offboard calibration method described here. https://docs.px4.io/master/en/advanced_config/sensor_thermal_calibration.html#offboard_calibration

I made offboard calibration, the result is the same, all logs and params are here 
https://we.tl/t-3FQiUReeao

Is it normal that gyto and acc ids are not in the same order/numbered
	Line 26: 1	1	CAL_**ACC0**_ID	1246218	6
	Line 41: 1	1	CAL_ACC_PRIME	1246218	6
	Line 855: 1	1	TC_**A1**_ID	1246218	6

dagar, 2020-04-22:
> I made offboard calibration, the result is the same, all logs and params are here
> https://we.tl/t-3FQiUReeao

For convenience: [wetransfer-c9996d.zip](https://github.com/PX4/Firmware/files/4516729/wetransfer-c9996d.zip)

dagar, 2020-04-22:
You have relatively huge offsets for the board rotation. Can you zero them and normal calibration (including board level horizon) again?

```
SENS_BOARD_X_OFF	12.146007537841796875
SENS_BOARD_Y_OFF	-13.161460876464843750
```

MartinStokelj, 2020-04-22:
Yes huge board offset is set after level horizon, level horizon is bad after calibrating accelometers.

MartinStokelj, 2020-04-23:
According to logs, the TC is probably added to wrong accelometer. 
LSM303D showing wrong offsets when board is on the table.

[image]
[image]
[image]

MartinStokelj, 2020-04-28:
Hi I am trying to test this new FW but I do not see TC_* params, I don't know how to enable TC now.

[image]

MartinStokelj, 2020-04-28:
After compiling px4_fmu-v2_default.px4 I saw that there is no TC_A_ENABLE param...
Reading the web I saw we need to build as px4_fmu-v3_default.px4 to get the TC params.
With new FW the calibration off acc,gyro was success, and after level horizon all was ok, so I think the problem is solved.

dagar, 2020-04-28:
Thanks for testing @MartinStokelj, but it's not technically resolved until https://github.com/PX4/Firmware/pull/14776 is fully reviewed and merged.

MartinStokelj, 2020-05-04:
Hi,
today I was able to create new thermal calibration files and put it on the drone. After calibration of sensors I was not able to ARM the drone. Error "High Accelometer Bias".
If I disabled the acc1 and set acc0 as prime, I was able to ARM the drone.
CAL_ACC1_ID	2162698
CAL_ACC_PRIME 1114122

I have upload the files (Vtol_ser18AP_thermal_cal.params is done after ARM error)
[TC_error.zip](https://github.com/PX4/Firmware/files/4573569/TC_error.zip)

dagar, 2020-05-04:
> Hi,
> today I was able to create new thermal calibration files and put it on the drone. After calibration of sensors I was not able to ARM the drone. Error "High Accelometer Bias".
> If I disabled the [...]

Can you attach a log from boot showing the attempted arming?

MartinStokelj, 2020-05-04:
Here it is
[debugConsole.txt](https://github.com/PX4/Firmware/files/4575161/debugConsole.txt)

MartinStokelj, 2020-05-05:
Here is ulg file, I thought you wanted console log...
[ulg.zip](https://github.com/PX4/Firmware/files/4579108/ulg.zip)

dagar, 2020-05-05:
> Here is ulg file, I thought you wanted console log...
> [ulg.zip](https://github.com/PX4/Firmware/files/4579108/ulg.zip)

That only has the raw sensor data (SDLOG_PROFILE). https://logs.px4.io/plot_app?log=f573f21d-edc0-4b32-8707-d12fb88da724

We need to see the regular logging to verify all the corrected accel data going into the estimator looks good.

MartinStokelj, 2020-05-06:
> 
> 
> > Here is ulg file, I thought you wanted console log...
> > [ulg.zip](https://github.com/PX4/Firmware/files/4579108/ulg.zip)
> 
> That only has the raw sensor data (SDLOG_PROFILE). https://logs.px4.io/plot_app?log=f573f21d-edc0-4b32-8707-d12fb88da724
> 
> We need to see the regular logging to verify all the corrected accel data going into the estimator looks good. If you're able to re-test with a full log please try current PX4 master (after #14784 merged as a fix for this).

I have used PX4 master, did the sensors calibration again and after that arming went ok.
https://review.px4.io/plot_app?log=7d5b9a30-1618-4822-acee-e0dd52ea0dd3

MartinStokelj, 2020-05-25:
Today we did 6 flights with VTOL and all went ok. Thermal calibration works, since we didn't had to calibrate sensors after each flight.

dagar, 2020-06-18:
Fixed.

closed 2020-06-18 by dagar