PX4/PX4-Autopilot #14719: NuttX STM32H7 SPI6 DMA not working

dagar, 2020-04-21:
This came up when testing the ICM20649 driver on the new CUAV X7Pro board (~ 37% cpu on wq:SPI6).

On STM32H743 SPI6 is on DMAMUX2 BDMA (D3 domain).

``` Console
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   0 Idle Task                   71425 30.863   244/  512   0 (  0)  READY  3
   1 hpwork                          0  0.000   344/ 1260 249 (249)  w:sig  3
   2 lpwork                          9  0.000  1096/ 1612  50 ( 50)  w:sig  3
   3 init                         1243  0.000  1848/ 2924 100 (100)  w:sem  3
   4 wq:manager                      0  0.000   384/ 1252 255 (255)  w:sem  4
 227 log_writer_file                 0  0.000   368/ 1164  60 ( 60)  w:sem  3
  17 dataman                        57  0.000   760/ 1204  90 ( 90)  w:sem  4
  19 wq:lp_default                  45  0.000   812/ 1700 205 (205)  READY  4
  31 wq:hp_default                2613  1.220  1004/ 1900 241 (241)  w:sem  4
  38 wq:SPI1                     10863  4.882  1928/ 2492 253 (253)  w:sem  4
  41 wq:SPI4                     28218 12.728  1020/ 2492 250 (250)  w:sem  4
  45 wq:SPI6                     82377 36.791  1912/ 2492 248 (248)  READY  4
  87 wq:att_pos_ctrl             12727  5.754  5024/ 7196 242 (242)  READY  4
  88 wq:rate_ctrl                 3284  1.482   992/ 1660 255 (255)  w:sem  4
  93 commander                    1040  0.435  1280/ 3212 140 (140)  w:sig  6
  94 commander_low_prio              2  0.000   716/ 2996  50 ( 50)  w:sem  6
 101 mavlink_if0                  4269  1.918  1832/ 2572 100 (100)  READY  4
 103 mavlink_rcv_if0               298  0.087  2664/ 3908 175 (175)  READY  4
 144 gps                            62  0.000  1016/ 1676 208 (208)  w:sem  4
 180 mavlink_if1                  1295  0.610  1832/ 2484 100 (100)  w:sig  4
 181 mavlink_rcv_if1               349  0.174  2512/ 3908 175 (175)  w:sem  4
 191 wq:UART5                      548  0.261   664/ 1396 234 (234)  w:sem  4
 200 navigator                     132  0.087  1000/ 1764 105 (105)  w:sem  5
 224 logger                        381  0.087  2184/ 3644 233 (233)  w:sem  3
 240 top                            26  1.133  1560/ 2028 248 (248)  RUN    3

Processes: 25 total, 7 running, 18 sleeping, max FDs: 15
CPU usage: 67.65% tasks, 1.48% sched, 30.86% idle
DMA Memory: 5120 total, 1024 used 1024 peak
Uptime: 224.165s total, 71.425s idle
```

``` Console
nsh> work_queue status

Work Queue: 8 threads                      RATE        INTERVAL
|__ 1) wq:lp_default   
|   |__ 1) send_event                   30.0 Hz      33325.6 us (33333 us)
|   \__ 2) load_mon                      1.0 Hz     995538.7 us (1000000 us)
|__ 2) wq:hp_default   
|   |__ 1) tone_alarm                   10.1 Hz      99264.0 us
|   |__ 2) rc_update                     0.0 Hz          0.0 us
|   |__ 3) adc                         100.0 Hz      10000.0 us (10000 us)
|   \__ 4) battery_status              100.0 Hz       9999.5 us (10000 us)
|__ 3) wq:SPI1         
|   |__ 1) rm3100                       97.6 Hz      10246.0 us
|   \__ 2) icm20689                   1000.4 Hz        999.6 us
|__ 4) wq:SPI4         
|   |__ 1) bmi088                     1666.5 Hz        600.1 us (600 us)
|   |__ 2) bmi088                     1249.9 Hz        800.1 us (800 us)
|   \__ 3) ms5611                       99.7 Hz      10033.3 us
|__ 5) wq:SPI6         
|   |__ 1) icm20649                    900.0 Hz       1111.1 us (1111 us)
|   \__ 2) ms5611                       99.4 Hz      10064.8 us
|__ 6) wq:att_pos_ctrl 
|   |__ 1) sensors                     250.1 Hz       3998.5 us
|   |__ 2) vehicle_acceleration       1000.3 Hz        999.7 us
|   |__ 3) vehicle_air_data             74.7 Hz      13380.6 us
|   |__ 4) vehicle_imu                 250.1 Hz       3998.4 us
|   |__ 5) vehicle_imu                 456.1 Hz       2192.5 us
|   |__ 6) vehicle_imu                 300.0 Hz       3333.4 us
|   \__ 7) ekf2                        250.1 Hz       3998.4 us
|__ 7) wq:rate_ctrl    
|   \__ 1) vehicle_angular_velocity   1000.4 Hz        999.6 us
\__ 8) wq:UART5        
    \__ 1) rc_input                    250.0 Hz       4000.1 us (4000 us)
```

dagar, 2020-04-21:
FYI @bkueng

davids5, 2020-06-09:
@dagar See https://github.com/PX4/Firmware/pull/15068

dagar, 2020-09-08:
Fixed properly in the NuttX upgrade. https://github.com/PX4/Firmware/pull/15139

closed 2020-09-24 by davids5