PX4/PX4-Autopilot #14721: Takeoff denied in HITL Simulation with Gazebo

LordHelmchen324, 2020-04-21:
**Describe the bug**
When running a HITL setup with Gazebo, I can arm (the rotors in Gazebo start spinning) but I cannot takeoff. QGC's error message is only "Takeoff command temporarily rejected". In the log it says "Critical: Takeoff denied! Please disarm and retry". Doing so leads to the same result.
A SITL setup with Gazebo works fine. Using JMavSim in a HITL setup works perfectly, too.
One difference I have noticed when comparing my error scenario to the two that work: QGC says the value of Alt (Rel) is NaN (or --.-- on the HUD).

**To Reproduce**
Steps to reproduce the behavior:
1. Install the development toolchain according to the [Toolchain Installation](https://dev.px4.io/v1.9.0/en/setup/dev_env.html).
2. Follow the [instructions for setting up a HITL simulation](https://dev.px4.io/v1.9.0/en/simulation/hitl.html) with Gazebo.
    a) In step 4 of the Simulation Setup for Gazebo I use `/dev/tty.usbmodem01` on my MacBook Pro. On an iMac I've had to use `/dev/tty.usbmodem1`.
    b) In a recent commit the iris world was removed, so instead of running Gazebo as in step 5 of the Gazebo Setup section, I've had to run `make px4_sitl gazebo` again to achieve the same result. I have already had the described problem before this change was made.
3. After starting QGC in step 6 of the instructions for the Gazebo Setup, arm the drone and select Takeoff in QGC.

**Expected behavior**
The drone should takeoff without errors and I should be able to see that it does so in the Gazebo client.

**Log Files and Screenshots**
Flight log file: [Log file link](https://review.px4.io/plot_app?log=629eabe0-cfbb-4cd8-ad61-29d150175b0f)

**Drone:**
I am using an FMUK66 board by NXP.

**Additional context**
I am running macOS 10.15.4.

julianoes, 2020-04-22:
[image]

Something is off with the baro data.

Which airframe did you select for the Gazebo HITL sim?

LordHelmchen324, 2020-04-22:
I am using the Iris model from the `sitl_gazebo` directory (the one you are supposed to modify in the [setup instructions](https://dev.px4.io/v1.9.0/en/simulation/hitl.html)).

LordHelmchen324, 2020-05-06:
I was using the pymavutil recently to listen to the port 14540 (API/Offboard) and because I continue to have this issue, I figured I could checkout how the messages differ from SITL to HITL (working vs non-working). I pretty much just started the simulation and then dumped 100 of the MAVLink messages sent to port 14540 using Python.

I found that the ALTITUDE messages in the HITL setup come with a lot of NaNs (there are none in the SITL setup). See for yourself below.

**SITL:**
`ALTITUDE {time_usec : 223864000, altitude_monotonic : 488.6708984375, altitude_amsl : 0.051495034247636795, altitude_local : 0.051495034247636795, altitude_relative : 0.026063019409775734, altitude_terrain : -0.0595221184194088, bottom_clearance : 0.1110171526670456}`

**HITL (which is broken):**
`ALTITUDE {time_usec : 193650085, altitude_monotonic : 31751.267578125, altitude_amsl : nan, altitude_local : nan, altitude_relative : nan, altitude_terrain : nan, bottom_clearance : nan}`

Looking at the HIGHRES_IMU messages you can see issues, too.

**SITL:**
`HIGHRES_IMU {time_usec : 223664000, xacc : -0.050586096942424774, yacc : -0.04609781503677368, zacc : -9.856425285339355, xgyro : -0.0003387816250324249, ygyro : -0.0026601599529385567, zgyro : -0.0027051102370023727, xmag : 0.010812134481966496, ymag : -0.2165789157152176, zmag : 0.4258551299571991, abs_pressure : 95596.15625, diff_pressure : 0.07217077165842056, pressure_alt : 488.262451171875, temperature : 11.82763671875, fields_updated : 8191}`

**HITL:**
`HIGHRES_IMU {time_usec : 193295350, xacc : nan, yacc : nan, zacc : nan, xgyro : nan, ygyro : nan, zgyro : nan, xmag : -38467371008.0, ymag : 4.577901953102745e-41, zmag : 2.8388660745900986e-15, abs_pressure : -140.41615295410156, diff_pressure : 0.0, pressure_alt : nan, temperature : 1344092416.0, fields_updated : 7167, id : 0}`

I'm not quite sure what that means as to where the issue comes from. I believe Gazebo runs mostly the same code when putting together barometer messages, regardless of SITL or HITL.

MaxorPaxor, 2020-07-12:
Have you figured it out @jank324 ? I'm having the same issue

jank324, 2020-07-12:
> Have you figured it out @jank324 ? I'm having the same issue

Unfortunately, I haven't.

I did poke around the Gazebo MAVLink interface plugin a while back, but I wasn't able to find anything wrong. (I'm not exactly familiar with PX4 source code though.) I'm hoping that this will be fixed with issue #14449, which is on the Release Blockers To Do list. But it seems no one got around to it yet.

In the meantime, I'm just using a Gazebo SITL setup. It's not ideal for my project, but otherwise works perfectly.

jungx148, 2020-08-05:
Having a exactly same problem with @jank324.

julianoes, 2021-04-29:
I expect this to be fixed with https://github.com/PX4/PX4-Autopilot/pull/17481.

kois86, 2021-09-09:
Has this issue been resolved? i have si

> > Have you figured it out @jank324 ? I'm having the same issue
> 
> Unfortunately, I haven't.
> 
> I did poke around the Gazebo MAVLink interface plugin a while back, but I wasn't able to find anything wrong. (I'm not exactly familiar with PX4 source code though.) I'm hoping that this will be fixed with #14449, which is on the Release Blockers To Do list. But it seems no one got around to it yet.
> 
> In the meantime, I'm just using a Gazebo SITL setup. It's not ideal for my project, but otherwise works perfectly.

Has this issue resolved? i have similar problem. I have even changed from linux to windows machine just to be able to change the flowcontrol, and enable, but the problem similar problem still persist.

thank you in advance for ur expert feedback