PX4/PX4-Autopilot #14722: GPS # of satellites used dropped to 16 (u-blox M9N)

limhyon, 2020-04-21:
**Describe the bug**
I was cherry-picking up 
https://github.com/PX4/GpsDrivers/commit/cf87129bd341eb56222117bf5b75c18e01ddd500

to test M9N module. But as soon as PX4 firmware boots and send few commands to M9N, the number of satellites used on M9N dropped to 16 (fixed number) from around 30~32 satellites.

**To Reproduce**
1. Connect u-center to M9N with USB port.
2. Wait PX4 booted.
3. Check u-center.

**Expected behavior**
It should have maximum number of satellites observed.

julianoes, 2020-04-22:
FYI @bkueng

ryanjAA, 2020-04-23:
Cherrypicked and same result. 16 sats Max. Tried a different firmware in same location at the same time and more than 16 sats so not hardware.

bkueng, 2020-04-24:
Apparently this is indeed a hardware restriction (in combination with how we configure it): with 10Hz update rate you only get up to 16 satellites. Reducing to 5Hz allows to go higher.

I reduced to 5Hz, so we can use more satellites: https://github.com/PX4/GpsDrivers/pull/57

ryanjAA, 2020-04-27:
Tested at 25hz and still limited to 16 sats. Tested at 5hz and no limitation so unfortunately it is a hardware constraint. https://github.com/PX4/GpsDrivers/pull/57 Breaks functionality with the M9N (as a few other things are added in the PR) but testing this with the previous driver change works fine with the reduced hz changes.

dagar, 2020-04-27:
> Tested at 25hz and still limited to 16 sats. Tested at 5hz and no limitation so unfortunately it is a hardware constraint. [PX4/GpsDrivers#57](https://github.com/PX4/GpsDrivers/pull/57) Breaks functionality with the M9N (as a few other things are added in the PR) but testing this with the previous driver change works fine with the reduced hz changes.

I'm just curious, are you gathering and comparing any other metrics along with the satellite count and output data rate?

ryanjAA, 2020-04-27:
What specifically? HDOP and VDOP? HDOP drops by half when going from 16 to 22-24 sats and when hitting 29+ starts to decrease further.

[image]
[image]

julianoes, 2020-07-27:
Fixed in https://github.com/PX4/GpsDrivers/pull/57, right?

bkueng, 2020-07-27:
Correct, thanks Julian.

closed 2020-07-27 by bkueng