PX4/PX4-Autopilot #14735: EKF2 velocity estimate wrong - v1.11-beta

mhkabir, 2020-04-22:
**Describe the bug**
We are flying in indoor environments with VIO, and while testing recently, discovered that the EKF's velocity estimate significantly differed from that provided by the vision system. We don't know if this is a new issue, or whether this already existed before, since we have not flown this setup so extensively before. We also only recently plumbed the vision system's velocity estimate through to PX4, so it's easy to confirm the issue now.

[image]

**Expected behavior**
The velocity estimate matches the one provided by the vision system, which is accurate and has been benchmarked against groundtruth. Also the vehicle was flying in a straight line, and that velocity offset wasn't actually there in real life.

**Log Files and Screenshots**
Here are two logs from boot illustrating the issue. The first one is without velocity fusion enabled, and the second one is with velocity fusion enabled (which doesn't make much of a difference).

1. EKF2_AID_MASK = 24 (ev pos, ev yaw): https://logs.px4.io/plot_app?log=6eb5a9cb-a4e9-4b4a-89eb-e71e11419670
2. EKF2_AID_MASK = 280 (ev pos, ev yaw, ev vel): https://logs.px4.io/plot_app?log=dcb3b53b-07d2-4db9-a118-592ef82d5cba

These are both replayable. Our branch is based on upstream `746b3124abe9df486c6d88a3d9ec0c438e63654d`

First log:
Accel bias estimate:
[image]

Second log
Accel bias estimate:
[image]

**Drone (please complete the following information):**
- 550 sized carbon quadcopter.
- Hardmounted Pixhawk 4.

dagar, 2020-04-23:
The accel (icm20649) calibration changed quite a bit. Notably there's no longer a huge x offset. This was also seen in https://github.com/PX4/Firmware/pull/14676#issuecomment-617512387. I'd say there was something bogus in the old mpu6000 driver. 

### Old
```
CAL_ACC0_XOFF, -1.0432920455932617
CAL_ACC0_XSCALE, 0.9367385506629944
CAL_ACC0_YOFF, 0.07577945291996002
CAL_ACC0_YSCALE, 0.9988186955451965
CAL_ACC0_ZOFF, -0.09410905838012695
CAL_ACC0_ZSCALE, 0.9868613481521606
```

### New
```
CAL_ACC0_XOFF, -0.052748680114746094
CAL_ACC0_XSCALE, 0.9987621903419495
CAL_ACC0_YOFF, 0.08501672744750977
CAL_ACC0_YSCALE, 0.9985091686248779
CAL_ACC0_ZOFF, -0.08153867721557617
CAL_ACC0_ZSCALE, 0.9868456125259399
```

LorenzMeier, 2020-04-23:
@dagar I would add code to wipe the params on update. The factory calibration of the ICM20649 is so good that you're better off having none.

mhkabir, 2020-04-23:
@RomanBapst @bresch @kamilritz @jkflying Can you have a look?

dagar, 2020-04-24:
The integrated data in the accel x axis looks questionable in sensor_combined.

[image]

[image]

This is where the estimated bias (~0.2 m/s) is coming from.

[image]

[image]

I've tried to replicate this on another fmu-v5, but no success so far. It seems like it could be a numerical problem.

mhkabir, 2020-04-28:
Happened again on master: https://logs.px4.io/plot_app?log=6a006545-b3bf-446f-9baf-f89c5fcfb542

Same vehicle, with @dagar's new fix attempts, and a fixed companion computer fan (so lower vibrations on ground)

[image]

bresch, 2020-04-28:
@mhkabir Your VIO system seems to provide a velocity estimate with a variance of 5 m2/s2
[image]

bresch, 2020-04-28:
@mhkabir Your VIO system seems to provide a velocity estimate with a variance of 5 m2/s2 (std dev: 2.2m/s). A typical GPS gives velocity aiding with an accuracy of 0.5m/s. [image]

kamilritz, 2020-04-28:
@bresch @mhkabir  Oh that is for sure not optimal. This remembers me of a bug that I wanted to fix inside the velocity-in-body-frame PR (, which is not merged yet). I am sorry for not creating a single PR for the bug fix. Can you spot the [bug](https://github.com/PX4/Firmware/blob/master/src/modules/ekf2/ekf2_main.cpp#L1126-L1128)?

mhkabir, 2020-04-28:
@bresch Good catch with the variances. What you're looking at is actually the _position_ variance, thanks to the above bug. This is fishy regardless (such a large position variance is not right), and seems to be something on our side, looking into it.

mhkabir, 2020-04-28:
There seems to have been a bunch of separate issues in play here:

mhkabir, 2020-04-28:
There seem to have been a bunch of issues in play simultaneously here:
- Accelerometer bias estimates were exploding, which seems to have been fixed thanks to @dagar (https://github.com/PX4/Firmware/pull/14752 and https://github.com/PX4/Firmware/pull/14765)
- Our vision position variances were wrong, found thanks to @bresch. We will fix this properly on our side longer term, for now using a fixed variance using `EKF2_NOISE_MD` seems to work just fine.
- Vision velocity fusion was using the wrong variances, which is fixed thanks to @kamilritz (https://github.com/PX4/Firmware/pull/14779)

Thanks for all the help everyone!

closed 2020-04-28 by mhkabir