PX4/PX4-Autopilot #14743: Fixed Wing Flaps channel 6 provides PWM output channel 7 not

jstrebel, 2020-04-23:
On my FW 1.10.1 I have the flaps on channel 6 & 7. All works fine.
I gave a try with 1.11.beta and observe that only the flap on channel 6 is moving. Channel 7 is dead. 
Reverted back to 1.10.1 both Flaps working again.
QGC is on 4.0.6
Please let me know If you need futher info

Mixer file is on ./microsd/etc/mixers

### AAERTWF.main.mix
#### Skywalker 2000
##  10.4 16:30
###
#### servo 1 Quer rechts
M: 1
O:      10000  10000      0 -10000  10000
S: 0 0  10000  10000      0 -10000  10000
## S: 0 6  10000  10000      0 -10000  10000
#### servo2 Quer links
M: 1
O:      10000 10000    0 -10000  10000
S: 0 0  10000  10000   0 -10000  10000
### S: 0 6 -10000 -10000     0 -10000  10000
### servo 3 Hoehe
M: 1
O:       10000   10000      0 -10000  10000
S: 0 1  -10000  -10000      0 -10000  10000
### servo4 Seite
M: 1
O:      10000  10000      0 -10000  10000
S: 0 2  10000  10000      0 -10000  10000
### servo 5 Motor
M: 1
O:      10000  10000      0 -10000  10000
S: 0 3      0  20000 -10000 -10000  10000
#####
##### zusaetze
### servo 6 flaps links
M: 1
O:      -10000  -10000      0 -10000  10000
S: 0 4  0  20000  -10000 -10000  10000
###
### servo 7 Flaps rechts
M: 1
O:      10000  10000      0 -10000  10000
S: 0 4  0  20000     -10000 -10000  10000
##### s8
M: 1
O:      10000  10000      0 -10000  10000
S: 0 4  10000  10000      0 -10000  10000
#####
M: 1
O:      10000  10000      0 -10000  10000
S: 0 4  10000  10000      0 -10000  10000

sfuhrer, 2020-04-25:
@jstrebel could you share a log from this? Even from some bench testing would be useful, so just arm the vehicle on the ground and apply the flaps. Thanks!

jstrebel, 2020-04-26:
the first is the log 1.10.1
https://review.px4.io/plot_app?log=23e498af-9db1-4794-9464-35c3a2bd529e
this is the log with to 1.11.beta. 
Hope this helps. ( I did downgrade to 1.10.1 und upgrade to 1.11.beta several times. NO change with the hardware nor parameter changes

sfuhrer, 2020-04-26:
Thanks for the logs. One thing I find weird is that there is also some output on the AUX, even though the aux mixer is set to pass. Not sure if that's related though.
[image]

sfuhrer, 2020-04-26:
@jstrebel could you maybe run a test with all the other stuff beside the settings for the Flap channels removed in the mixer, so only:

servo 6 flaps links
M: 1
O: -10000 -10000 0 -10000 10000
S: 0 4 0 20000 -10000 -10000 10000

servo 7 Flaps rechts

M: 1
O: 10000 10000 0 -10000 10000
S: 0 4 0 20000 -10000 -10000 10000

?

LorenzMeier, 2020-04-27:
@jstrebel I would test removing all comment lines. Multi-line comments and alike things are not officially supported in mixers.

jstrebel, 2020-04-27:
@LorenzMeier, I deleted the comment lines. No change in behavior.
@sfuhrer If I put only the Flap entries in the mixer file, then as expected the two ailerons (channel 1 and 2) move as flaps in the correct direction.
- If I put the two entries of the ailerons as the first two entries, then the ailerons work correct. The Flaps move now at channel 3 and 4 (yaw and elevon)
In case I go back to the complete mixer File with all channels then the Flaps problem is back. 
I do not see any strange character in the mixer file. 
Strange to me is the identical mixer file works correct with 1.10.1 but not in 1.11.beta. Should I upload the mixer file?

jstrebel, 2020-04-28:
@sfuhrer, did you get the required info you requested from me?
Is there a reason that this issue is not marked as a bug?

sfuhrer, 2020-04-28:
@dagar do you have any idea what could have changed on the mixer level lately?

jstrebel, 2020-04-28:
I made some further experiments. If I put the following mixer file on the sd card then I get:
#### AAERTWF.main.mix
#### Skywalker 2000
##  10.4 16:30
#### servo 1 Quer rechts
M: 1
O:      10000  10000      0 -10000  10000
S: 0 0  10000  10000      0 -10000  10000
#### servo2 Quer links
M: 1
O:      10000 10000    0 -10000  10000
S: 0 0  10000  10000   0 -10000  10000
### servo 3 Hoehe
M: 1
O:       10000   10000      0 -10000  10000
S: 0 1  -10000  -10000      0 -10000  10000
### servo4 Seite
M: 1
O:      10000  10000      0 -10000  10000
S: 0 2  10000  10000      0 -10000  10000
### servo 6 flaps links
M: 1
O:      -10000  -10000      0 -10000  10000
S: 0 4  0  20000  -10000 -10000  10000
### servo 7 Flaps rechts
M: 1
O:      10000  10000      0 -10000  10000
S: 0 4  0  20000  -10000 -10000  10000
Servo 1&2 roll working correct
Servo 3&4 pitch and yaw woring correct
Servo 5&6 Flaps working correct

**Test with the test mixer file:**
If I put duplicate now channel 4 (channel 4&5 now the same mixer setting. Servo 5 moves like servo 4, as expected) 
Flap on channel 6 work ok but Flap on channel 7 does not move.

#### AAERTWF.main.mix
#### Skywalker 2000
##  10.4 16:30
#### servo 1 Quer rechts
M: 1
O:      10000  10000      0 -10000  10000
S: 0 0  10000  10000      0 -10000  10000
#### servo2 Quer links
M: 1
O:      10000 10000    0 -10000  10000
S: 0 0  10000  10000   0 -10000  10000
### servo 3 Hoehe
M: 1
O:       10000   10000      0 -10000  10000
S: 0 1  -10000  -10000      0 -10000  10000
### servo4 Seite
M: 1
O:      10000  10000      0 -10000  10000
S: 0 2  10000  10000      0 -10000  10000
### servo4 Seite
M: 1
O:      10000  10000      0 -10000  10000
S: 0 2  10000  10000      0 -10000  10000
### servo 6 flaps links
M: 1
O:      -10000  -10000      0 -10000  10000
S: 0 4  0  20000  -10000 -10000  10000
### servo 7 Flaps rechts
M: 1
O:      10000  10000      0 -10000  10000
S: 0 4  0  20000  -10000 -10000  10000

jstrebel, 2020-04-29:
Observation: If I remove the sd card, forcing px4 starts with the built in mixer (Bormatec Maya) channel 4 Yaw and channel 6 Flaps show the same value if I do pwm info.

jstrebel, 2020-04-29:
Observation: If I remove the sd card, forcing px4 starts with the built in mixer (Bormatec Maya) channel 4 Yaw and channel 6  in this case wheel ( show the same value if I do pwm info). There is no pwm output value on channel 7  
[image]

RomanBapst, 2020-04-30:
@jstrebel Really weird. I assume that you are using IO for pwm, correct?

jstrebel, 2020-04-30:
Correct I/O PWM OUT on the pixhawk4. Its really weird. Going back to 1.10.1 everything works again.

RomanBapst, 2020-04-30:
@jstrebel Can you try to use the pwm test command on that channel?

jstrebel, 2020-04-30:
I did the following test with the standard Bormatec mixer no SD CARD inserted
pwm test -c 6 -p 1000 and 1800 
pwm test -c 7 -p 1000 and 1800 
Result: Both Flap Servo move

ryanjAA, 2020-05-01:
I’ll load up a fresh plane with 11 beta (and the Albatross standard mixer) and report back the results on the flaps. We have flaps on ch 7 and 8.

jstrebel, 2020-05-02:
Just did another test putting my mixer file back to the SD Card.
pwm test -c 6 -p 1000 (and 1800)
pwm test -c 7 -p 1000 (and 1800)
Result: Both Flap Servos move
But stil no change via RC: only Flap on Ch 6 move

ryanjAA, 2020-05-02:
I loaded up 11 beta 1 and tested as well. We have 2 channel flap output, main 7 and 8 but triggered by RC channel 9 when done manually. It also doesn't work in our case. Looking at it a little closer, we have no output on anything from output 6 and up (servo_output_raw) shows this as well.

Taking from above, running pwm test -c 7 -p 1000 and we have movement but otherwise output shows 0 us.

mcsauder, 2020-05-04:
Hi @jstrebel and @ryanjAA.  If you add the following to your airframe config files does it change the behavior? 
`set PWM_OUT 12345678`

jstrebel, 2020-05-04:
@mcsauder : No change

mcsauder, 2020-05-05:
@jstrebel can confirm with his tests, but [this change](https://github.com/PX4/Firmware/commit/ca81175b076af83b361db581fb52a86675227f8e#diff-4b78c3b7d0adaf26c7dcb653513e5e9fL57) appears to have made this issue show up.

Hi @bkueng, you might be able to explain what the intention of this change was and if there is a better way to address the current behavior observed in this issue.

(@jstrebel , in the meantime, if you want, you can explicitly place `set PWM_OUT fmu` in your aircraft config file as well and that should take care of things without modifying the code.)

dagar, 2020-05-05:
> Hi @dagar , you might be able to explain what the intention of this change was and if there is a better way to address the current behavior observed in this issue.

I don't think I'm following the issue here. That change shouldn't have been more than a rename, is that not the case?

I'll take this issue and try to reproduce.

jstrebel, 2020-05-05:
@dagar, happy to help.

mcsauder, 2020-05-06:
@dagar , this issue appears fixed in current master but not in v1.11.0-beta1 as reported back by @jstrebel .... I'll poke around on this a bit more tomorrow.

jstrebel, 2020-06-11:
Just tested it on Beta2. Its ok now

closed 2020-06-11 by jstrebel