PX4/PX4-Autopilot #14744: SHA256 broken for px4.py

rblilja, 2020-04-23:
SHA256 broken for px4.py when installing tool chain using brew on Mac OS 10.14.6. 

> brew install px4-dev

==> Installing px4-dev from px4/px4
==> Downloading https://raw.githubusercontent.com/PX4/Firmware/master/Tools/px4.
######################################################################## 100.0%
Error: An exception occurred within a child process:
  ChecksumMismatchError: SHA256 mismatch
Expected: 48199ee9ff392eff8a1efcd177e2c10f4a4dfd9877e52e13b1f3540d5dfedac9
  Actual: dafe6a32d270c48cbd42960adb2155bd5f60d1aec8cb644b10f8cafc0c8d2e8f
 Archive: /Users/sven/Library/Caches/Homebrew/downloads/7be7daca52fab83c40ed326bdcf732c0060cdc0b9d97e0d94f1e71856982d00b--px4.py
To retry an incomplete download, remove the file above.

I have tried to remove, untap, cleanup and upgrade homebrew without success. Help needed. Thanks!

dagar, 2020-04-23:
@TSC21 do you know anything about this?

TSC21, 2020-04-23:
No. The only thing I did was updating the file output. One fix I saw regarding this is manually delete the file and run the brew update and brew install cmds again.

TSC21, 2020-04-23:
@rblilja did you have the toolchain before?

rblilja, 2020-04-23:
@TSC21 No this is a clean install. Removing the file does not help. Tried update commands, doctor says everything is OK.

TSC21, 2020-04-23:
> @TSC21 No this is a clean install. Removing the file does not help. Tried update commands, doctor says everything is OK.

Ok I think I know the solution. Standbye

TSC21, 2020-04-23:
@rblilja @dagar fix in https://github.com/PX4/homebrew-px4/pull/54.

TSC21, 2020-04-23:
@rblilja https://github.com/PX4/homebrew-px4/pull/54 was merged. Would you mind try again?

rblilja, 2020-04-23:
Success! Thank you!

closed 2020-04-23 by rblilja

LordHelmchen324, 2020-05-04:
I'm getting the exact same issue (seems to be even the same checksums) trying to install px4-sim. Does the checksum need to be updated for that recipe, too?

robinasu, 2020-06-03:
Hello, re-installing the dev toolchain on MacOS 10.15.5 on 6/2/2020 resulted in some errors, similar to the above. Are there any quick work-arounds?

Command `brew install px4-dev` led to the following issue :

`==> Downloading https://raw.githubusercontent.com/PX4/Firmware/master/Tools/px4.
Already downloaded: /Users/robinsimpson 1/Library/Caches/Homebrew/downloads/7be7daca52fab83c40ed326bdcf732c0060cdc0b9d97e0d94f1e71856982d00b--px4.py
Error: SHA256 mismatch
Expected: dafe6a32d270c48cbd42960adb2155bd5f60d1aec8cb644b10f8cafc0c8d2e8f
  Actual: 48199ee9ff392eff8a1efcd177e2c10f4a4dfd9877e52e13b1f3540d5dfedac9
 Archive: /Users/robinsimpson 1/Library/Caches/Homebrew/downloads/7be7daca52fab83c40ed326bdcf732c0060cdc0b9d97e0d94f1e71856982d00b--px4.py
To retry an incomplete download, remove the file above.
`

Command `brew install px4-sim` results in:

`==> Downloading https://raw.githubusercontent.com/PX4/Firmware/master/Tools/px4.
Already downloaded: /Users/robinsimpson 1/Library/Caches/Homebrew/downloads/7be7daca52fab83c40ed326bdcf732c0060cdc0b9d97e0d94f1e71856982d00b--px4.py
Error: SHA256 mismatch
Expected: dafe6a32d270c48cbd42960adb2155bd5f60d1aec8cb644b10f8cafc0c8d2e8f
  Actual: 48199ee9ff392eff8a1efcd177e2c10f4a4dfd9877e52e13b1f3540d5dfedac9
 Archive: /Users/robinsimpson 1/Library/Caches/Homebrew/downloads/7be7daca52fab83c40ed326bdcf732c0060cdc0b9d97e0d94f1e71856982d00b--px4.py
To retry an incomplete download, remove the file above.`

Command `sudo -H pip install pyserial empy toml numpy pandas jinja2 pyyaml` resulted in:

`Error: invalid option: -H`