PX4/PX4-Autopilot #14745: pixhawk mini IMU missing v1.11.0-beta1

dagar, 2020-04-24:
Two reports of the 3DR/Holybro pixhawk mini IMU not starting with v1.11.0-beta, however the single unit I have seems to be working.

dagar, 2020-04-24:
@davids5 do you have a unit you could check quickly?

bys1123, 2020-04-24:
Here is some screenshots
[image]
[image]
[image]

dagar, 2020-04-24:
Thanks @bys1123. Could you capture the output of `ver all` as well?

bys1123, 2020-04-24:
@dagar No problem. It's lunch time in china, will reply you half hour later.

bys1123, 2020-04-24:
[image]
@dagar

bys1123, 2020-04-24:
[image]

dagar, 2020-04-24:
Ok, I was able to get mine stuck in the same state after upgrading (automatic reboot). After a fresh power cycle (disconnect + reconnect usb) it works.

``` Console
FMUv2 ver 0xA : Rev 0 V2M
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V3
HW type: V2M
HW version: 0x0009000A
HW revision: 0x00000000
FW git-hash: 31f3a21849e45ea5f110ddb6de2b09a630f54fc6
FW version: 1.11.0 80 (17498240)
FW git-branch: master
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: 7fffab1610d2ea08a95383febd0579e0037b6b51
Build datetime: Apr 24 2020 00:10:24
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 000100000000313237313036510500470037
MCU: STM32F42x, rev. 3
[hardfault_log] Fault Log is Armed
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
nsh: rgbled_pwm: command not found
Board sensors: /etc/init.d/rc.board_sensors
rgbled #0 on I2C bus 2
ms5611 #0 on SPI bus 1 (devid=0x3d)
icm20608g #0 on SPI bus 1 (devid=0x3a)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
No autostart ID found
INFO  [logger] logger started (mode=all)
```

``` Console
nsh> work_queue status

Work Queue: 6 threads                      RATE        INTERVAL
|__ 1) wq:lp_default   
|   |__ 1) send_event                   30.0 Hz      33317.7 us (33333 us)
|   \__ 2) load_mon                      1.0 Hz     988905.0 us (1000000 us)
|__ 2) wq:hp_default   
|   |__ 1) tone_alarm                   10.2 Hz      98369.9 us
|   |__ 2) rc_update                     0.0 Hz          0.0 us
|   |__ 3) adc                         100.0 Hz       9998.7 us (10000 us)
|   \__ 4) battery_status              100.0 Hz       9998.6 us (10000 us)
|__ 3) wq:I2C2         
|   \__ 1) rgbled                       38.9 Hz      25734.8 us
|__ 4) wq:SPI1         
|   |__ 1) ms5611                       99.4 Hz      10063.4 us
|   \__ 2) icm20608g                   400.4 Hz       2497.7 us
|__ 5) wq:att_pos_ctrl 
|   |__ 1) sensors                     400.4 Hz       2497.6 us
|   |__ 2) vehicle_acceleration        400.4 Hz       2497.5 us
|   |__ 3) vehicle_air_data             74.5 Hz      13421.8 us
|   |__ 4) vehicle_imu                 400.4 Hz       2497.3 us
|   \__ 5) ekf2                        400.4 Hz       2497.4 us
\__ 6) wq:rate_ctrl    
    \__ 1) vehicle_angular_velocity    400.4 Hz       2497.5 us
```

bys1123, 2020-04-24:
@dagar Tried a fresh power cycle, still not works.

bys1123, 2020-04-24:
[image]

dagar, 2020-06-01:
I'm not seeing any issue with current master as long as px4_fmu-v3 is flashed (bootloader update may be required).

[image]

[image]

bys1123, 2020-06-01:
Thanks, we will try it again.
As you see, the board is already using FMUV3 bootloader, we need to flash it again for testing, right?

dagar, 2020-06-02:
> Thanks, we will try it again.
> As you see, the board is already using FMUV3 bootloader, we need to flash it again for testing, right?

Yes, please flash current master.

If it's still not working we'll probably need more info.
  - When did it last work? Log available?
  - Was there a board revision at some point?

dagar, 2020-06-02:
@bys1123 if master doesn't work please try https://github.com/PX4/Firmware/pull/15013.

bys1123, 2020-06-02:
@dagar Thanks! master doesn't work, will try #15013 tomorrow.

davids5, 2020-06-02:
master works for me with V3 build on the device. V2 does not have the driver.

master: (rebased under my branch)

V2 Build
```
FMUv2 ver 0xA : Rev 0 V2M
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V2
HW type: V2M
HW version: 0x0009000A
HW revision: 0x00000000
FW git-hash: 72034abe943e8390bf86e256ef9e91f46f08fc32
FW version: 1.11.0 80 (17498240)
FW git-branch: master_px4_fmu-v6x
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: 81d5b634cbc890ace6ad31c798d97d5112eee21f
Build datetime: Jun  2 2020 08:32:49
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 0001000000003634343035355106004e0027
MCU: STM32F42x, rev. 3
[hardfault_log] Fault Log is Armed
INFO  [param] selected parameter default file /fs/mtd_params
ERROR [parameters] ignoring unrecognised parameter 'BAT_CNT_V_CURR'
ERROR [parameters] ignoring unrecognised parameter 'BAT_CNT_V_VOLT'
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
  IMU_GYRO_RATEMAX: curr: 0 -> new: 400
nsh: send_event: command not found
nsh: rgbled_ncp5623c: command not found
nsh: rgbled_pwm: command not found
ERROR [param] Parameter LIGHT_EN_BLINKM not found
WARN  [px4io] check CRC failed: -22
INFO  [tune_control] Publishing standard tune 18
INFO  [px4io] default PWM output device
[PX4IO] using firmware from /etc/extras/px4_io-v2_default.bin
[PX4IO] found bootloader revision: 4
[PX4IO] erase...
[PX4IO] programming 58480 bytes...
[PX4IO] verify...
[PX4IO] update complete
INFO  [tune_control] Stopping playback...
INFO  [tune_control] Publishing standard tune 19
Board sensors: /etc/init.d/rc.board_sensors
rgbled #0 on I2C bus 2
hmc5883 #0 on I2C bus 1 (external, equal to '-b 1')
nsh: lis3mdl: command not found
nsh: ist8310: command not found
nsh: qmc5883: command not found
nsh: rm3100: command not found
ms5611 #0 on SPI bus 1 (devid=0x3d)
nsh: icm20608g: command not found
nsh: mpu9250: command not found
ERROR [param] Parameter TRIG_MODE not found
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
ERROR [param] Parameter DSHOT_CONFIG not found
INFO  [px4io] default PWM output device
INFO  [init] Mixer: /etc/mixers/quad_+.main.mix on /dev/pwm_output0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
ERROR [param] Parameter TC_A_ENABLE not found
ERROR [param] Parameter TC_B_ENABLE not found
ERROR [param] Parameter TC_G_ENABLE not found
ERROR [param] Parameter SENS_EN_PX4FLOW not found
[... 138 lines cut ...]
|   |__ 3) adc                           100.0 Hz         9999 us (10000 us)
|   |__ 4) battery_status                100.0 Hz         9999 us (10000 us)
|   \__ 5) pwm_out                         0.1 Hz      5478855 us
|__ 3) wq:I2C2
|   \__ 1) rgbled                         38.6 Hz        25888 us
|__ 4) wq:I2C1
|   \__ 1) hmc5883                       125.7 Hz         7956 us
|__ 5) wq:SPI1
|   |__ 1) ms5611                         99.4 Hz        10062 us
|   \__ 2) icm20608g                     400.3 Hz         2498 us
|__ 6) wq:navigation_and_controllers
|   |__ 1) sensors                       199.6 Hz         5009 us
|   |__ 2) vehicle_acceleration          222.6 Hz         4492 us
|   |__ 3) vehicle_air_data               74.3 Hz        13463 us
|   |__ 4) vehicle_imu                   202.4 Hz         4941 us
|   |__ 5) ekf2                          200.3 Hz         4993 us
|   |__ 6) mc_pos_control                 50.1 Hz        19956 us
|   \__ 7) land_detector                  50.0 Hz        19987 us (20000 us)
|__ 7) wq:rate_ctrl
|   |__ 1) vehicle_angular_velocity      400.6 Hz         2496 us
|   \__ 2) mc_rate_control               400.5 Hz         2497 us
\__ 8) wq:attitude_ctrl
    \__ 1) mc_att_control                200.3 Hz         4992 us
nsh>
```

bys1123, 2020-06-02:
[image]
Thanks for testing, We sure we are using V3 bootloader

davids5, 2020-06-02:
Is this just one unit or on many?

bys1123, 2020-06-02:
@davids5 many, include Holybro current produce Pixhawk Mini.

davids5, 2020-06-02:
@bys1123 can you get one sent to me?

bys1123, 2020-06-02:
@davids5 No problem, I'll let Holybro send you one.

bys1123, 2020-06-03:
Fixed by #15013

dagar, 2020-06-03:
Thanks for the quick feedback.

closed 2020-06-03 by dagar