PX4/PX4-Autopilot #14747: px4_sitl gazebo not working on ubuntu16.04 with gazebo9

Prasheel24, 2020-04-24:
Please suggest a work around for the pc hangup when run the command
```
make px4_sitl gazebo
```
-- computer freezes when doing this -- I have to press the power button to restart.

I am trying to simulate a plane in gazebo and work with the mavsdk libraries.
If this issue is still new, then I do have a pixhawk with which I can start HITL directly. But would need the steps to run mavsdk c++ file to run a mission on QGC and Xplane10.

Would be grateful to whoever is able to help me here.!

Thanks,
Prasheel

Prasheel24, 2020-04-24:
Update: 
So, I had been SMH for quite a few days on this, and finally got it to work!
I commented every dependency in the CMakeLists of sitl_gazebo. Somehow with all at once, the plugins do not work. I worked them out one by one.
But, there are some serious issues in usv_dynamics plugin as thats the last part in the commented part with which the issue still exists!. Also, had to first build on px4_sitl_default gazebo then worked it out on gazebo_plane. This finally got the model running with mission commands from QGC. 
But, did give these errors-
```
In file included from /usr/include/eigen3/Eigen/Core:316:0,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /home/prasheel/Workspace/RA/Dependencies/Firmware/Tools/sitl_gazebo/include/common.h:26,
                 from /home/prasheel/Workspace/RA/Dependencies/Firmware/Tools/sitl_gazebo/include/gazebo_opticalflow_mockup_plugin.h:37,
                 from /home/prasheel/Workspace/RA/Dependencies/Firmware/Tools/sitl_gazebo/src/gazebo_opticalflow_mockup_plugin.cpp:27:
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:60:39: warning: ignoring attributes on template argument ‘__m128 {aka __vector(4) float}’ [-Wignored-attributes]
 template<> struct is_arithmetic<__m128>  { enum { value = true }; };
                                       ^
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:61:40: warning: ignoring attributes on template argument ‘__m128i {aka __vector(2) long long int}’ [-Wignored-attributes]
 template<> struct is_arithmetic<__m128i> { enum { value = true }; };
                                        ^
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:62:40: warning: ignoring attributes on template argument ‘__m128d {aka __vector(2) double}’ [-Wignored-attributes]
 template<> struct is_arithmetic<__m128d> { enum { value = true }; };
                                        ^
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:160:43: warning: ignoring attributes on template argument ‘Eigen::internal::Packet4f {aka __vector(4) float}’ [-Wignored-attributes]
 template<> struct unpacket_traits<Packet4f> { typedef float  type; enum {size=4, alignment=Aligned16}; typedef Packet4f half; };
                                           ^
In file included from /usr/include/eigen3/Eigen/Core:316:0,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /home/prasheel/Workspace/RA/Dependencies/Firmware/Tools/sitl_gazebo/include/common.h:26,
                 from /home/prasheel/Workspace/RA/Dependencies/Firmware/Tools/sitl_gazebo/include/gazebo_opticalflow_mockup_plugin.h:37,
                 from /home/prasheel/Workspace/RA/Dependencies/Firmware/Tools/sitl_gazebo/src/gazebo_opticalflow_mockup_plugin.cpp:27:
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:161:43: warning: ignoring attributes on template argument ‘Eigen::internal::Packet2d {aka __vector(2) double}’ [-Wignored-attributes]
 template<> struct unpacket_traits<Packet2d> { typedef double type; enum {size=2, alignment=Aligned16}; typedef Packet2d half; };
                                           ^
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:162:43: warning: ignoring attributes on template argument ‘Eigen::internal::Packet4i {aka __vector(2) long long int}’ [-Wignored-attributes]
 template<> struct unpacket_traits<Packet4i> { typedef int    type; enum {size=4, alignment=Aligned16}; typedef Packet4i half; };
                                           ^
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:468:37: warning: ignoring attributes on template argument ‘Eigen::internal::Packet4f {aka __vector(4) float}’ [-Wignored-attributes]
 struct protate_impl<offset, Packet4f>
                                     ^
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:476:37: warning: ignoring attributes on template argument ‘Eigen::internal::Packet4i {aka __vector(2) long long int}’ [-Wignored-attributes]
 struct protate_impl<offset, Packet4i>
                                     ^
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:484:37: warning: ignoring attributes on template argument ‘Eigen::internal::Packet2d {aka __vector(2) double}’ [-Wignored-attributes]
 struct protate_impl<offset, Packet2d>
                                     ^
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:755:35: warning: ignoring attributes on template argument ‘Eigen::internal::Packet4f {aka __vector(4) float}’ [-Wignored-attributes]
 struct palign_impl<Offset,Packet4f>
                                   ^
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:778:35: warning: ignoring attributes on template argument ‘Eigen::internal::Packet4i {aka __vector(2) long long int}’ [-Wignored-attributes]
 struct palign_impl<Offset,Packet4i>
                                   ^
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:801:35: warning: ignoring attributes on template argument ‘Eigen::internal::Packet2d {aka __vector(2) double}’ [-Wignored-attributes]
 struct palign_impl<Offset,Packet2d>
                                   ^
/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:815:34: warning: ignoring attributes on template argument ‘Eigen::internal::Packet4f {aka __vector(4) float}’ [-Wignored-attributes]
 ptranspose(PacketBlock<Packet4f,4>& kernel) {
                                  ^
[... cut ...]
I believe there are some issues with the build as it does not let me take all these plugins at once. I specially upgraded to gazebo9 for this (compromised my work on turtlebot - doesnt work on g9). The memory taken up at runtime is huge that causes the machine to freeze? I would be happy to work it out with you guys! But for now the issue is resolved (except the error log from eigen, which disappears on successive runs). I hope there are no dependencies on the usv_dynamics plugin.

Please do let me know how I can be of any help guys!.

Thanks, 
Prasheel

julianoes, 2020-04-24:
> -- computer freezes when doing this -- I have to press the power button to restart.

I guess it's your computer running out of RAM during linking.

> cc1plus: warning: unrecognized command line option ‘-Wno-address-of-packed-member

This is a warning, so you should be able to get around it by disabling `-Werror`.

Prasheel24, 2020-04-24:
I am using an 8GB RAM with >50GB free space on my hard drive. I have an NVIDIA GPU. 
I would be happy to know the requirements to run the firmware?

bys1123, 2020-04-24:
RAM usage depends on your CPU thread (compiler jobs), usually need 2GB per job.

Prasheel24, 2020-04-24:
Well, I wonder how this same firmware works well on a Pixhawk2 (currently using for HITL). 
Please mind my knowledge in the field, kind of a rookie in flying drones but I'm good with C++!

Anyway, great to know about this. Would definitely take a look at the source code.

Thank you for the insight.

julianoes, 2020-06-04:
With #14971 you can now build with fewer cores in order to avoid running out of RAM:

```
make px4_sitl gazebo -2
```

closed 2020-06-04 by julianoes

cjffly, 2021-06-02:
> With #14971 you can now build with fewer cores in order to avoid running out of RAM:
> 
> ```
> make px4_sitl gazebo -2
> ```

I am not sure I should open another issue or just reply here: I have the same issue with gazebo11 on Ubuntu 20.04. Seems make here does not recognize '-2'. 
 ```
/PX4-Autopilot$ make px4_sitl gazebo -2
make: invalid option -- '2'
Usage: make [options] [target] ...
 ```

My PC has rayzon 5900, which is 12 core, with 16gb RAM. (Side question: should I upgrade to 32gb?)