PX4/PX4-Autopilot #14760: [MAVSDK TEST] Groundtruth always zero for vision and flow model

kamilritz, 2020-04-25:
Inside the mavsdk test, when using the offboard mode, ground truth information from the gazebo simulation is used to perform checks.
Apparently for the two models iris_vision, iris_opt_flow_mockup the ground truth information is not received correctly and is always zero. Therefore all tests are passing. 
[image]

The ground truth information seems correct for the plain iris (GPS) model.

How to check:
Print out `target_pos` and `current_pos` variable [here](https://github.com/PX4/Firmware/blob/master/test/mavsdk_tests/autopilot_tester.cpp#L252-L267) 

@julianoes I am pretty sure that this was not the case when I wrote those tests for the first time. Do you have any ideas what could have changed.

julianoes, 2020-04-27:
Thanks @kamilritz. I was seeing these errors too but then gave up debugging it.

1. We should not send 0, right!?
2. Why is the ground truth 0 for these models? That's also wrong, right?

kamilritz, 2020-04-27:
@julianoes 
1) I was printing out the ground truth inside the gazebo_gps_plugin. I never observed that it was generating a 0 in the beginning and switching to another value in the beginning. A hacky way to solve this, would be to check for an non zero ground truth position during the `request_ground_truth()` call. But actually a ground truth position of 0 could happen. Especially if it is not received correctly for models like in 2)
Could it be that somewhere inside the Firmware or MAVSDK a variable that got initialized to 0 is passed further.
2) I have no idea. I checked the uorb topic while running normal simulation with these models, and I could see the *_groundtruth topics as well. But I haven't check their content.

kamilritz, 2020-05-01:
@julianoes I was able to solve both issues with one change. It is caused by filling the GROUND_TRUTH message with content from the vehicle_global_position message. This message is coming from the estimator and is only published when the ekf is accepting global information. This is not the case for the iris_flow and iris_vision model. On the other side the ekf is publishing the vehicle_global_position message only after the gps data is passing the checks. This is causing the ground truth message to show 0 in the beginning until the ekf is publishing the information.

This can be easily solved by using to the raw gps data for the ground truth. I will open a PR.

closed 2020-05-02 by kamilritz

julianoes, 2020-05-04:
Thanks @kamilritz!