PX4/PX4-Autopilot #14762: ERROR: cannot launch node of type [px4/px4]: can't locate node [px4] in package [px4]

pkyadav73199, 2020-04-26:
**Describe the bug**
ERROR: cannot launch node of type [px4/px4]: can't locate node [px4] in package [px4]

I am getting the above error while launching gazebo7 on ubuntu 16.04, drone has been launched in gazebo environment but i am not able to connect with FCU while runing with mavros node.

kamilritz, 2020-04-26:
Did you build the gazebo simulation? By `DONT_RUN=1 make px4_sitl gazebo`

pkyadav73199, 2020-04-27:
YES  I did.

I have launched the gazebo by running the following codes.

DONT_RUN=1 make px4_sitl_default gazebo
source ~/catkin_ws/devel/setup.bash    # (optional)
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 posix_sitl.launch

kamilritz, 2020-04-27:
Then I don't know what could be the issue, sry. I am sure, someone else can.

0b10010010, 2020-06-25:
Hello,

I was getting similar error along with permission error and I was able to mitigate the issue by running the scripts in: https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html

Hope this helps

rsj007, 2022-02-25:
I also met the problem. The reason is that PX4 is not installed successfully. My error is libgazebo-dev failed. I installed gazebo9 in ubuntu16.04 and solved.