PX4/PX4-Autopilot #14775: ninja: build stopped: subcommand failed

pkyadav73199, 2020-04-27:
**Describe the bug**
I am trying to simulate UAV using px4 with mavros node but i am getting error while launching gazebo.

ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

**To Reproduce**
mkdir -p ~/src
cd ~/src
git clone https://github.com/PX4/Firmware.git
cd Firmware
git submodule update --init --recursive
make px4_sitl gazebo

**Expected behavior**
I am getting the error after running the above code.

julianoes, 2020-04-27:
There should be more error than that. Can you paste all the output? And also try a clean build?

pkyadav73199, 2020-04-27:
-- PX4 version: v1.11.0-beta1-93-g70329ce
-- PX4 config file: /home/edith/src/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-267-g8a9d961
-- Configuring done
-- Generating done
-- Build files have been written to: /home/edith/src/Firmware/build/px4_sitl_default
[7/745] git submodule src/drivers/gps/devices
[10/745] Generating mixer_multirotor.generated.h
FAILED: cd /home/edith/src/Firmware/build/px4_sitl_default/src/lib/mixer/MultirotorMixer && /usr/bin/python3 /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py -f /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_x.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_deadcat.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_h.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_plus.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_s250aq.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_vtail.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_wide.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x_cw.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x_pusher.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_y.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/tri_y.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/twin_engine.toml -o mixer_multirotor.generated.h
Failed to import toml: No module named 'toml'

You may need to install it using:
    pip3 install --user toml

[10/745] Generating mixer_multirotor_6dof.generated.h
FAILED: cd /home/edith/src/Firmware/build/px4_sitl_default/src/lib/mixer/MultirotorMixer && /usr/bin/python3 /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py --sixdof -f /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_x.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_deadcat.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_h.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_plus.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_s250aq.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_vtail.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_wide.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x_cw.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x_pusher.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_y.toml /home/edith/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/tri_y.toml /home/edith/src/Firmware/src/lib/mixer [... cut ...] ders.msg wind_estimate.msg yaw_estimator_status.msg -i /home/edith/src/Firmware/msg -o /home/edith/src/Firmware/build/px4_sitl_default/uORB/topics -e templates/uorb -t /home/edith/src/Firmware/build/px4_sitl_default/msg/tmp/headers -q
Failed to import em: No module named 'em'

You may need to install it using:
    pip3 install --user empy

[10/745] Generating git version header
ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

mhkabir, 2020-04-27:
The compile error output above tells you what to do, please read it.

closed 2020-04-27 by mhkabir

julianoes, 2020-04-28:
```
pip3 install --user empy
```

pkyadav73199, 2020-04-28:
Thanks a lot .
It works.
I have done simulation successfully .