PX4/PX4-Autopilot #14777: v1.11 reset icm20689 accel scale factor if ~ 2

dagar, 2020-04-28:
With the new icm20689 the scale factor comes out quite close to 1. Existing setups calibrated with PX4 v1.10 (the mpu6000 driver) seem to have a scale factor on one axis that's nearly 2.

This is a bit hacky, but at param import time we can identify this in the parameters and reset the scale factor if it exists only on one axis and exceeds something close to 2.

FYI @MaEtUgR

MaEtUgR, 2020-04-28:
@dagar I didn't see a scale of 2 in my cases. Is that a different issue?

I was reporting the case where I had to recalibrate the accelerometer on both vehicles that had a Firmware older than 2 months ago and I updated to master. I recalibrated because both vehicles showed a steep pitch angle when completely level in the attitude estimate represented by QGC's virtual horizon after updating.

The quadrotor uses a pixhawk 4 mini
The plane uses a pixhawk 4

| | quad before | quad after | plane before | plane after |
|---|---|---|---|---|
| CAL_ACC0_ID | 3932170 | 3932170 | 3866634 | 3932170 |
| CAL_ACC0_XOFF | 1.581990 | -0.173167 | -2.507170 | -0.041409 |
| CAL_ACC0_XSCALE | 0.932632 | 1.001169 | 0.925561 | 0.999273 |
| CAL_ACC0_YOFF | 0.150241 | 0.251485 | 0.071872 | 0.041645 |
| CAL_ACC0_YSCALE | 0.997267 | 0.999228 | 1.003336 | 0.985294 |
| CAL_ACC0_ZOFF | -0.238199 | -0.227139 | -0.043277 | 0.089027 |
| CAL_ACC0_ZSCALE | 0.987909 | 0.985522 | 0.984737 | 0.978124 |

These are always the primary accelerometers and what I see is a clear change in x-axis offset which would also explain the pitch error. I do not know how the same calibration was considered perfectly level before since I flew both platforms a lot without issues or need to recalibrate the accel.

As a solution to the offset problem I only see recalibration. Maybe it would have not even armed because of consistency checks but I didn't dare to try.

dagar, 2020-04-28:
The root cause appears to be this undocumented register applied to ICM devices in the old mpu6000 driver. This mpu6000 driver was used for the mpu6000, icm20602, icm20608g, icm20689 up until recently.

https://github.com/PX4/Firmware/blob/ccaa103164a09d66ff9faf4fcdfb39a09ba834a1/src/drivers/imu/mpu6000/MPU6000.cpp#L194

dagar, 2020-04-28:
I think the only question then is if we force recalibration of everything if the offset is this bad or can we get away with only clearing the single offset under the right conditions.

bys1123, 2020-04-28:
Maybe we should add a parameter storage what version firmware been used in previous bootup. Then with some mapping work the program could know what initialize progress it should do after firmware update.

bys1123, 2020-04-28:
If not, I guess we would reset recalibration everytime after firmware update in the future?

bys1123, 2020-05-02:
After upgrade to 1.11, pre-flight check will tell users accel inconsistent, so may not cause un-safety?

dagar, 2020-05-02:
> After upgrade to 1.11, pre-flight check will tell users accel inconsistent, so may not cause un-safety?

Yes, so this is more about trying to make the upgrade experience less unpleasant.

bys1123, 2020-05-03:
I flash back to 1.10, shows accel is uncalibrated.

dagar, 2020-06-18:
I've decided to leave it alone. If the offset is sufficiently different and beyond the estimator bias limits you'll be unable to arm until you recalibrate.

This is a good reminder that we should be a bit more discerning before letting undocumented magic fixes slip into the code base. https://github.com/PX4/Firmware/blob/6afa7e4368c4bb925b298f2f50aa3c50cdebdf71/src/drivers/imu/mpu6000/MPU6000.cpp#L666-L669

closed 2020-06-18 by dagar