PX4/PX4-Autopilot #14783: SITL Z-Drift in position estimate when drone is doing nothing

shubang93, 2020-04-28:
I'm trying to use PX4 SITL using the Typhoon H480 model. All I do is instantiate the simulator using the following commands: 
```
DONT_RUN=1 make px4_sitl gazebo_hexa_x_tilt     
source /opt/ros/melodic/setup.bash     
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default     
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)     
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo     
roslaunch px4 mavros_posix_sitl.launch vehicle:=typhoon_h480 world:=/workspace/Firmware/Tools/sitl_gazebo/worlds/typhoon_h480.world verbose:=true
```
Then, I just let it sit there for about 200 seconds. A couple of things happen: 

1. There is a sudden increase in noise along Z for IMU measurement
2. Eventually there is a shift in the average value of IMU - Z measurement
At the second point (circled in the plot), there is a large jump in position estimate of Z. In all of this, the GPS altitude measurement is very stable. What is causing this issue? Am I doing something wrong? 
[image]

The plots from QGroundControl are below: 
[image]

To reiterate, I'm doing absolutely nothing. Not even arming the vehicle.

julianoes, 2020-04-29:
Interesting, thanks for the issue.
FYI @Jaeyoung-Lim @bresch

Jaeyoung-Lim, 2020-05-02:
@shubang93 I think this might be related to the oscillations in the vehicle. Can you maybe see if this is reproducable on the iris?

Jaeyoung-Lim, 2020-05-21:
Related to https://github.com/PX4/Firmware/issues/14953

Jaeyoung-Lim, 2021-01-17:
Fixed in https://github.com/PX4/PX4-Autopilot/pull/16519

closed 2021-01-17 by Jaeyoung-Lim