PX4/PX4-Autopilot #14787: pixhawk is not allowing GPS to be fixed

mhk4zemi, 2020-04-29:
Hi guys

Recently I have an issue with my flight controller. I can see on the Qgroundcontrol that GPS recognizes many satellites for instance 10 or more, although Pixhawk seems cannot sense it so it is not reading the position of the drone. The value of the HDOP and VDOP of the GPS signal is low.  I have also uploaded a picture of the noise and jamming level of the GPS.

This is happening once we have a power tether attached to the drone close to GPS, I am not sure if I can reduce the noise. 

[image]
[image]
[image]

I also increase the minimum acceptable values in the EK2_REQ_* section but it did not help at all. 

Any comments would be much appreciated. :)

julianoes, 2020-04-29:
Could you share the link to the log?

ryanjAA, 2020-05-01:
What’s contained in the power tether? Just power or are communications on the same bundle of wires? Also, what is the power type? Ac or dc? And if ac is there and ac converter onboard the plane and if dc is it high voltage with an onboard step down, etc. we have a bit of experience with high voltage power tethers and switching frequency and quite a few things can cause havoc. Definitely a log but more detailed info on your setup would be helpful from a diagnostics perspective.

mhk4zemi, 2020-05-01:
@julianoes Can you please tell me what part you are interested to know about the log file?

mhk4zemi, 2020-05-01:
@ryanjAA . Tether line is just carrying DC power and we have dc converter onboard as well. If you can tell me what part is interesting I can take it out from the log file and share it here.

ryanjAA, 2020-05-01:
It’s a little more involved than that. Generally if you’re seeking help you should share as much detail as possible and definitely at a minimum the entire log if you’d like someone to look at it as there is often things you might not see as important that someone else might spot the problem. 

You have an atypical setup which yields more to understanding what you have onboard and what you’re doing to try and help. The log is the best first step. I’d detail what I have onboard as well given you could possibly be sending up 400-600v dc and then bringing it down back to 25 or 50v for onboard motor use which means understanding your hardware and power is a huge area of concern given you’re saying the gps issues arise when connecting the power tether. 

How are creating the voltage up and down? Are they isolated? Etc etc, there are a lot of questions but rather than questions it’s easier to get help if you just provide whatever information you can about your setup and that aside which often is r necessary, a log is. I’d turn logging on from boot and turn it all on then plug in the tether and share that log too (any log that has both cases). 

We used vicor units to bring 500v dc back down to something useable and I can tell you, there is a lot to consider and shield when doing this...

mhk4zemi, 2020-05-01:
@ryanjAA @julianoes 

Here is the complete logfile. In this one, pixhawk locked the GPS but in the middle of the hovering phase.  You can see it in the log data as well. The tether voltage is 400 and we have converters on board to take it down to 36.  We did not use any special shielding for the autopilot or the GPS antenna. 

https://review.px4.io/plot_app?log=5eb72abd-8794-4f31-b985-b695c1da4b16

julianoes, 2020-05-04:
From the log it looks indeed like the GPS accuracy is not very "good", and not "good enough". The estimated accuracy is around 4 meters when I would expect values around 1-2 meters max.

mhk4zemi, 2020-05-04:
> From the log it looks indeed like the GPS accuracy is not very "good", and not "good enough". The estimated accuracy is around 4 meters when I would expect values around 1-2 meters max.

How should I allow pixhawk to lock the GPS even with this amount of uncertainty?  Do you know what parameter should be changed?

ryanjAA, 2020-05-04:
The main easy and topical things you can do are going to be:

Move the physical GPS unit

Try a different unit (something with an Active antenna might help or add a larger ground plane just to increase the amount of sats)

Shield the unit and if possible measure surrounding EMF.

It looks like you are also using a custom version of PX4 so not sure what else you may have changed but unlikely that is hurting raw GPS reception. What's the highest amount of sats and lowest HDOP/VDOP you've been able to attain in your area? With and without your tether?

If your sat amount is high and hdop/vdop low until you plug in the tether then you know the issue. If you have a low baseline in terms of gps accuracy already (when the tether is not plugged in) then it is getting swamped from the tether itself and likely the fast voltage switching, if that's not the case then i'd look at the physical unit or its placement as well (of the GPS and wiring). Did you make or buy the DC-DC converters?

Also, are you using any onboard video? Specifically in the 1.3ghz range? That effects GPS a lot as well if the transmitter is too close and/or not good quality.

Also, you're having a lot of log drop outs, try changing the SD card (or format it if its already a high quality one).