PX4/PX4-Autopilot #14792: Port new airship vehicle type from v1.10 to master

daniel-leonard-robinson, 2020-04-29:
As mentioned in [https://discuss.px4.io/t/new-vehicle-type-airship/16514](https://discuss.px4.io/t/new-vehicle-type-airship/16514), our team has created an airship vehicle type and the next step is to port it to master from v1.10 before it can be integrated.

It would be helpful if any caveats or challenges can be highlighted here. Once it is ported, the relevant documentation can also be updated!

dagar, 2020-04-29:
I'm not sure about specific caveats or challenges in general, but depending on how far along you are I'd encourage you to open smaller incremental pull if you'd actually like detailed feedback.

As for maintaining support for a new vehicle type like this (that not many of us have access to) it will be quite helpful if you can add some basic automated testing (eg simple mission in gazebo).

dan-leo, 2020-05-07:
I've ported the airship to master, and submitted a PR in [sitl_gazebo](https://github.com/PX4/sitl_gazebo/pull/490)!

closed 2020-05-10 by dan-leo