PX4/PX4-Autopilot #14799: FMUv4 fails to build (with uavcan error)

tstellanova, 2020-04-30:
- Clone PX4/Firmware master branch (or stable, doesn't make a difference)
- Build for FMU v4 fails:

```
todd@skylab px4_fw %  make px4_fmu-v4_default  
[0/1317] git submodule mavlink/include/mavlink/v2.0
[3/1317] Running dsdl compiler
FAILED: libuavcan_dsdlc_run.stamp 
cd /Users/todd/Documents/proj/px4_fw/src/drivers/uavcan/libuavcan/libuavcan && /usr/local/bin/python3 /Users/todd/Documents/proj/px4_fw/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc test/dsdl_test/root_ns_a test/dsdl_test/root_ns_b /Users/todd/Documents/proj/px4_fw/src/drivers/uavcan/libuavcan/libuavcan/../dsdl/uavcan -Oinclude/dsdlc_generated && /usr/local/Cellar/cmake/3.15.3/bin/cmake -E touch /Users/todd/Documents/proj/px4_fw/build/px4_fmu-v4_default/libuavcan_dsdlc_run.stamp
Traceback (most recent call last):
  File "/Users/todd/Documents/proj/px4_fw/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc", line 59, in <module>
    from libuavcan_dsdl_compiler import run as dsdlc_run
  File "/Users/todd/Documents/proj/px4_fw/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py", line 18, in <module>
    from uavcan import dsdl
ModuleNotFoundError: No module named 'uavcan'
[4/1317] PX4 UAVCAN dsdl compiler
FAILED: src/drivers/uavcan/px4_uavcan_dsdlc_run.stamp 
cd /Users/todd/Documents/proj/px4_fw/build/px4_fmu-v4_default/src/drivers/uavcan && /usr/local/bin/python3 /Users/todd/Documents/proj/px4_fw/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc /Users/todd/Documents/proj/px4_fw/src/drivers/uavcan/dsdl/com /Users/todd/Documents/proj/px4_fw/src/drivers/uavcan/libuavcan/dsdl/uavcan -O/Users/todd/Documents/proj/px4_fw/build/px4_fmu-v4_default/src/drivers/uavcan/include/dsdlc_generated && /usr/local/Cellar/cmake/3.15.3/bin/cmake -E touch px4_uavcan_dsdlc_run.stamp
Traceback (most recent call last):
  File "/Users/todd/Documents/proj/px4_fw/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc", line 59, in <module>
    from libuavcan_dsdl_compiler import run as dsdlc_run
  File "/Users/todd/Documents/proj/px4_fw/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py", line 18, in <module>
    from uavcan import dsdl
ModuleNotFoundError: No module named 'uavcan'
[6/1317] Copying NuttX compressed config nsh and inflating (make olddefconfig)
ninja: build stopped: subcommand failed.
make: *** [px4_fmu-v4_default] Error 1
todd@skylab px4_fw % 
```

- I'm using a Mac OSX build machine that successfully builds eg Durandal.

dagar, 2020-04-30:
> I'm using a Mac OSX build machine that successfully builds eg Durandal.

Durandal doesn't have UAVCAN. 

Python version is my first guess. We have simple Mac OS CI builds that are still working. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-compile_mac/detail/master/1834/pipeline/13

Which python interpreter is found in the cmake configure?

tstellanova, 2020-05-01:
Looks like it finds python 3.7.7 during the build:

```
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.7.7", minimum required is "3") 
...
-- Found PythonInterp: /usr/local/bin/python3 (found version "3.7.7") 
```

I did not realize Durandal isn't using uavcan. I confirmed the same problem occurs when building FMUv5.

tstellanova, 2020-05-01:
If I just clone the [libuavcan submodule](https://github.com/PX4/libuavcan) I'm able to build that library just fine.  There's something about this px4 build step that breaks:
```
[1/1321] Running dsdl compiler
FAILED: libuavcan_dsdlc_run.stamp 
```

dagar, 2020-05-01:
That's a relatively new edition where we now manually generate in tree DSDL (https://github.com/PX4/Firmware/tree/master/src/drivers/uavcan/dsdl/com/hex/equipment/flow).

https://github.com/PX4/Firmware/blob/c5341da8137f460c84f47f0e38293667ea69a6cb/src/drivers/uavcan/CMakeLists.txt#L99-L103

tstellanova, 2020-05-01:
@avionicsanonymous I’m not sure what your build environment looks like, but do these slightly older px4 build targets work for you?

avionicsanonymous, 2020-05-01:
> 
> 
> @avionicsanonymous I’m not sure what your build environment looks like, but do these slightly older px4 build targets work for you?

I tried px4_fmu-v4_default and it builds fine for me. Running an Ubuntu 14.04 VM setup pretty much per the PX4 dev guide.

dagar, 2020-05-01:
> If I just clone the libuavcan submodule I'm able to build that library just fine. There's something about this px4 build step that breaks:

Are you able to see if it's defaulting to python (python 2)?

tstellanova, 2020-05-02:
Upgrading cmake from 3.15 to 3.17 apparently fixes this problem.  
Thanks for the help...closing.

closed 2020-05-02 by tstellanova