PX4/PX4-Autopilot #14802: Frsky Telemetry not working on Holybro Kakute F7

benjinne, 2020-04-30:
When trying to use any of the UARTs on the Kakute, and configuring TEL_FRSKY_CONFIG to the matching tx pin **no sensors** (other than rxbat and rssi) are discovered on my transmitter (a Jumpter T16). 

**To Reproduce**
Steps to reproduce the behavior:
1. Set TEL_FRSKY_CONFIG
2. Discover new Sensors on transmitter

**Expected behavior**
A list of sensors should be displayed on the transmitter

**Other steps I tried*
1. using the non-inverted smart-port pad on my r-xsr
2. verifying packets are being sent using "frsky_telemetry status"
3. changing the mode using "frsky_telemetry -m" to both sport and sport_single

dagar, 2020-05-01:
Any thoughts on this @bkueng?

bkueng, 2020-05-04:
Yes it requires NuttX changes to make this work on F7 - I'm still waiting on @davids5 for testing it.

justas-, 2020-05-14:
Affects Kakute F7 AIO as well.

benjinne, 2020-05-14:
> Affects Kakute F7 AIO as well.

That's what I'm using

bys1123, 2020-06-24:
@bkueng Is this F7 NuttX change required include fmuv5?

justas-, 2020-06-30:
Are there any updates on this?

davids5, 2020-06-30:
Sorry - I am running behind on getting to it.

davids5, 2020-08-08:
The fix for this is un https://github.com/PX4/Firmware/pull/15492 that was merged. If this is not resolved please re-open

closed 2020-08-08 by davids5